mantidproject/mantid #28050: Open Interfaces unexpected exception

DanielMurphy22, 2020-02-24:
**Original reporter:** Found by @Harrietbrown 

### Expected behavior
Should be able to open each interface
### Actual behavior

### Steps to reproduce the behavior
Open Plot or Workbench, open any of the following interfaces:
- Diffraction>HFIR 4Circle
Following error trace:
```
Traceback (most recent call last):
  File "C:\MantidNightlyInstall\bin\workbench\app\mainwindow.py", line 417, in <lambda>
    action.triggered.connect(lambda checked_py, script=script: self.launch_custom_python_gui(script))
  File "C:\MantidNightlyInstall\bin\workbench\app\mainwindow.py", line 377, in launch_custom_python_gui
    self.interface_executor.execute(open(filename).read(), filename)
  File "C:\MantidNightlyInstall\bin\mantidqt\widgets\codeeditor\execution.py", line 155, in execute
    exec (code_obj, self.globals_ns, self.globals_ns)
  File "C:/MantidNightlyInstall/scripts/HFIR_4Circle_Reduction.py", line 12, in <module>
    from HFIR_4Circle_Reduction import reduce4circleGUI  # noqa
  File "C:\MantidNightlyInstall\scripts\HFIR_4Circle_Reduction\reduce4circleGUI.py", line 34, in <module>
    from HFIR_4Circle_Reduction import PreprocessWindow
  File "C:\MantidNightlyInstall\scripts\HFIR_4Circle_Reduction\PreprocessWindow.py", line 10, in <module>
    import reduce4circleControl
ModuleNotFoundError: No module named 'reduce4circleControl'

- Direct>DGSPlanner
```
Traceback (most recent call last):
  File "C:\MantidNightlyInstall\bin\workbench\app\mainwindow.py", line 417, in <lambda>
    action.triggered.connect(lambda checked_py, script=script: self.launch_custom_python_gui(script))
  File "C:\MantidNightlyInstall\bin\workbench\app\mainwindow.py", line 377, in launch_custom_python_gui
    self.interface_executor.execute(open(filename).read(), filename)
  File "C:\MantidNightlyInstall\bin\mantidqt\widgets\codeeditor\execution.py", line 155, in execute
    exec (code_obj, self.globals_ns, self.globals_ns)
  File "C:/MantidNightlyInstall/scripts/DGSPlanner.py", line 14, in <module>
    planner = DGSPlannerGUI.DGSPlannerGUI()
  File "C:\MantidNightlyInstall\scripts\DGSPlanner\DGSPlannerGUI.py", line 55, in __init__
    self.instrumentWidget = InstrumentSetupWidget.InstrumentSetupWidget(self)
  File "C:\MantidNightlyInstall\scripts\DGSPlanner\InstrumentSetupWidget.py", line 220, in __init__
    self.updateFigure()
  File "C:\MantidNightlyInstall\scripts\DGSPlanner\InstrumentSetupWidget.py", line 274, in updateFigure
    self.gonfig.set_aspect('equal', adjustable='datalim')
  File "C:\MantidNightlyInstall\bin\lib\site-packages\matplotlib\axes\_base.py", line 1280, in set_aspect
    raise NotImplementedError(
NotImplementedError: It is not currently possible to manually set the aspect on 3D axes
```
-Utility>QECoverage
```
Traceback (most recent call last):
  File "C:\MantidNightlyInstall\bin\workbench\app\mainwindow.py", line 417, in <lambda>
    action.triggered.connect(lambda checked_py, script=script: self.launch_custom_python_gui(script))
  File "C:\MantidNightlyInstall\bin\workbench\app\mainwindow.py", line 377, in launch_custom_python_gui
    self.interface_executor.execute(open(filename).read(), filename)
  File "C:\MantidNightlyInstall\bin\mantidqt\widgets\codeeditor\execution.py", line 153, in execute
    code_obj = compile(block.code_str, filename, mode=COMPILE_MODE,
  File "C:/MantidNightlyInstall/scripts/QECoverage.py", line 461
    def indirect_input_check(self, ana):
    ^
IndentationError: unexpected indent
```
(might this last one be fixed by #28048 ?)
QEcoverage DOES open in MantidPlot...
### Platforms affected
Tested on 4.2.20200211.2150 Nightly Workbench and Plot

martyngigg, 2020-02-24:
If it helps the breakage of HFIR 4Circle is because of Python 3. Lines 10, 11 & 13 of  [PreprocessWindow.py](https://github.com/mantidproject/mantid/blob/master/scripts/HFIR_4Circle_Reduction/PreprocessWindow.py#L10) are missing the HFIR_4Circle_Reduction package prefix line line 12.

martyngigg, 2020-02-24:
I suspect QECoverage to be fixed by #28048 

The `DGSPlanner` problem must be with matplotlib 3.1 I guess. Did this work on Windows before?

martyngigg, 2020-02-24:
I think this is another set of system tests we want to add during maintenance: try to start every interface along with the workbench itself as mentioned in #28048.

martyngigg, 2020-02-24:
I found matplotlib/matplotlib#/13474 which looks like the call never did the right thing so they've made it raise an error. A suggestion is to make the figure itself square if it is required but at minimum just removing the call should be enough.

DanielMurphy22, 2020-02-25:
> I suspect QECoverage to be fixed by #28048
> 
> The `DGSPlanner` problem must be with matplotlib 3.1 I guess. Did this work on Windows before?

`DGSPlanner` opened fine in 4.2 and in Nightly on 131219 so presumeably its related to the matplotlib upgrade

PR #28065 (2020-02-25): Fix Interface Opening Exceptions
Testing for this PR should be done on *at least* windows and one other OS before being approved. 

**Description of work.**
 - Fixed imports in HFIR Diffraction GUI
- Removed `set_aspect` call on 3d plot which never worked properly on 3d plots and was removed from matplotlib 3.1 and was stopping the Direct DGS planner opening. 
- QECoverage didn't seem to require changes and may have been fixed by a previous PR. 
 
**To test:**
1. Check all interfaces open, particularly:
a. Diffraction -> HFIR 4Circle
b. Direct -> DGSPlanner
c. Utility -> QECoverage

Fixes #280 [... cut]
closed 2020-02-25 by martyngigg