mantidproject/mantid #28054: Mantid Plot crashes when trying to fit BivariateGaussian

Harrietbrown, 2020-02-24:
if you try to perform a fit with the function `BivariateGaussian` in Mantid Plot it will crash with the error message:
```
Error in execution of algorithm Fit:
cannot reshape array of size 2684 into shape (50,50,2)
  at line 99 in 'C:/MantidNightlyInstall/plugins/python/functions\BivariateGaussian.py'
```

If you try to fit with this function in `Multi dataset fitting` it will not crash Plot but the error message will still be raised.

alicerussell1, 2020-02-27:
I have looked into this and the exception is not being caught by the try catch in the `fitpropertybrowser` but its not obvious why. As this has not (yet) been noticed by a user and was in previous releases, I am moving this to do after the release.

alicerussell1, 2020-08-26:
This appears to be caused by the same issue as #28058 so closing as duplicate.

closed 2020-08-26 by alicerussell1