mantidproject/mantid #28058: Fitting browser crashes mantid if a fit is perfomed using PeakHKLErrors

Harrietbrown, 2020-02-24:
If a fit is performed using the `PeakHKLErrors` function it will crash mantid with the following error:
```
Error in execution of algorithm Fit:
Unable to find Data Object type with name '': data service  search object
```
1. Load MAR11060
2. Plot spectra 0
3. try to fit `PeakHKLErrors`.

alicerussell1, 2020-02-27:
This crash looks to be caused by the same thing as #28054.

NickDraper, 2020-04-20:
1 day timebox for further investigation

alicerussell1, 2020-08-26:
From investigation I found that on a release build of mantid (plot or workbench) it would hard crash if any function threw an exception during it's execution (not when declaring properties). This should be caught by the try catch statement in the [fitpropertybrowser](https://github.com/mantidproject/mantid/blob/master/qt/widgets/common/src/FitPropertyBrowser.cpp#L1699) but I cannot find a reason why it isn't. 
Also I found that fairly often it would catch the exception correctly the first time a fit was run but then it would crash the second time you did the fit.
When the same fit is run from a script it catches the exception correctly every time and does not crash.

This also appears to be a windows only crash.

As this has not been raised by a user and has been in many previous releases it is not worth putting more time into it so I am going to close this.

closed 2020-08-26 by alicerussell1