mantidproject/mantid #28059: Multi-dataset fitting fit some functions because it doesn't recognise some paramiters

Harrietbrown, 2020-02-24:
This error is found in several functions but the first ones I found were `Peak->Gaussian`, `Background->CubicSpline`, `Calibrate->ProductQuadraticExp`. There may be more with this issue.

1. Load some data
2. Open `Multi-dataset fitting`
3. Load the data
4. Add the function `Peak->Gaussian`
5. Click fit
6. The fit will not be performed and `Multi-dataset fitting` status will say `function doesn't depend on paramiter A0`

alicerussell1, 2020-03-02:
I've looked into this and I only get this error message when I set the minimizer to `Damped GaussNewton` or `LevenbergMarquardtMD` and don't change the parameters. This error makes sense in that case as it calculates the Hessian and if it is 0 for that parameter it will give the error you see. Do you see this issue in any other situation? (If you want to see the Hessian output change the log level to debug and it will print it when the fit is done)

Harrietbrown, 2020-03-02:
Yes you're right it does only return the error for those minimizers. If it is the case that this is the expected behavior for these minimizers the details on the status feedback could be more helpful in reflecting that.

closed 2021-03-08 by stale[bot]