mantidproject/mantid #28072: Segfault/ hard crash doesn't bring up error reporter

DanielMurphy22, 2020-02-25:
Found in unscripted testing #28045 
### Expected behavior
Should get an error reporter after a hard crash
### Actual behavior
Don't!
### Steps to reproduce the behavior
Open MantidPlot, run the algorithm segfault after unticking DryRun,
There is no error reporter!
### Platforms affected
May be windows only...

ConorMFinn, 2020-02-25:
I couldn't replicate this on the Ubuntu 18.04 nightly

NickDraper, 2020-02-25:
This is reproducible on windows mantidplot nightly

dtasev, 2020-02-26:
The error reporter after segfault should be launched by this https://github.com/mantidproject/mantid/blob/master/buildconfig/CMake/Packaging/launch_workbench.ps1#L49-L53, if this is only reproducible on Windows then that might be a good place to start looking.

dtasev, 2020-02-26:
If the `if %errorlevel% NEQ 0` line is commented out so that the error reporter is always started, this can be reproduced faster (just exit mantid), and gives me the following trace:
```
Traceback (most recent call last):
  File "C:\MantidNightlyInstall\bin\lib\site-packages\qtpy\__init__.py", line 204, in <module>
    from PySide import __version__ as PYSIDE_VERSION  # analysis:ignore
ModuleNotFoundError: No module named 'PySide'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\MantidNightlyInstall\bin\lib\site-packages\qtpy\__init__.py", line 210, in <module>
    raise PythonQtError('No Qt bindings could be found')
qtpy.PythonQtError: No Qt bindings could be found
```
the environment variables look alright
```
PATH=/c/MantidNightlyInstall/bin:/c/MantidNightlyInstall/plugins/qt4:<... other things I have>
PV_PLUGIN_PATH=C:\MantidNightlyInstall\plugins\paraview\qt4
QT_API=pyqt4
_BIN_DIR=C:\MantidNightlyInstall\bin
_EXTRA_PATH_DIRS=C:\MantidNightlyInstall\bin;C:\MantidNightlyInstall\plugins\qt4
_INSTALL_DIR=C:\MantidNightlyInstall
```
I can, however, `import qtpy` inside the MantidPlot script editor
```
[1] import qtpy

[2] ?qtpy

Type:        module
String form: <module 'qtpy' from 'C:\\MantidNightlyInstall\\bin\\lib\\site-packages\\qtpy\\__init__.py'>
File:        c:\mantidnightlyinstall\bin\lib\site-packages\qtpy\__init__.py
```

martyngigg, 2020-02-26:
I think I've found the problem. In [qtpy/__init__.py](https://github.com/mantidproject/thirdparty-msvc2015/blob/master/lib/python3.8/Lib/site-packages/qtpy/__init__.py#L179) it imports `sip` to set the relevant api. The latest versions of PyQt have a private sip module such as `PyQt4.sip` so the import fails and it skips to trying PySide. I think we can get around this by fiddling with PYTHONPATH when starting the error reporter. It's only an issue for MantidPlot because PyQt5 doesn't do the `sip` import.

dtasev, 2020-02-26:
👍 That fixes it.

martyngigg, 2020-02-26:
Thanks for confirming @dtasev.

PR #28116 (2020-02-27): Update PYTHONPATH for windows error reporter launch
**Description of work.**

As part of Python 3 migration `sip` got installed inside `PyQt4` on Windows. MantidPlot fixes up the Python path so this isn't an issue but the error reporter startup needs to do the same.

**To test:**

To test properly you will need to build a package. You could also just take the changes and modify and existing `launch_mantidplot.bat` in the currently nightly and see it works...

Fixes #28072 

*This does not require release notes* because **broken this dev cycle**

---

#### Reviewer ####

Please comment on the f [... cut]
closed 2020-02-27 by gvardany