mantidproject/mantid #28074: Sample logs bug

AnthonyLim23, 2020-02-25:
From spencer

https://forum.mantidproject.org/t/workbench-show-sample-logs-not-working/650

AnthonyLim23, 2020-02-25:
Nightly build of 25/2/20…
I get the following error message

Could not open sample logs for workspace ‘osiris120227_multi_graphite002_iqt’.
Could not open sample logs for workspace ‘osiris120227_multi_graphite002_iqt’.
Traceback (most recent call last):
File “C:\MantidNightlyInstall\bin\mantidqt\widgets\samplelogs\presenter.py”, line 87, in plot_logs
to_plot = [row for row in self.view.get_selected_row_indexes()
File “C:\MantidNightlyInstall\bin\mantidqt\widgets\samplelogs\view.py”, line 148, in get_selected_row_indexes
return [row.row() for row in self.table.selectionModel().selectedRows()]
AttributeError: ‘NoneType’ object has no attribute ‘selectedRows’

DanielMurphy22, 2020-02-25:
To reproduce download OSIRIS120227 from the archive and show the sample logs. Then untick the releative time box, get the error code above

martyngigg, 2020-02-25:
So it looks like this is not an issue for all files. It's only an issue where some text that is part of a log: name or the units uses a non-utf8 encoded character. In this case the unit on the `phase10` and `phase6` logs is `0xb5s`. `0xb5` is a UTF-16 encoded μ but we are trying to decode using `utf-8`. MantidPlot happened to get this right because `QString` uses [UTF-16 internally](https://wiki.qt.io/QString) and we weren't doing any translation. I'm checking with Freddie if there is anything that tells us what the encoding might be. Otherwise we might have to try a couple and then fail gracefully.

PR #28095 (2020-02-26): Sample Logs: Catch exception reading log values/metadata
**Description of work.**

See commit for more detail.

**Report to:** Spencer @ ISIS

**To test:**

Use the file specified in the issue and try to display the a sample logs in workbench. If should now warn on 2 logs giving a unicode decoding error. 

Fixes #28074 

*This does not require release notes* because **it's a minor issue with string encodings**

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of  [... cut]
closed 2020-02-26 by peterfpeterson