mantidproject/mantid #28090: Additional 'none' item in the legend of 1D plots

StephenSmith25, 2020-02-25:
If the legend of a 1D plot is modified on Ubuntu an extra 'none' entry will appear in the legend. 

Its possible the legend needs to be redrawn using ax.legend() followed by fig.canvas.draw(). For instance, the following steps remove the extra entry. 

(1) Plot the data and add errors, and then go to the IPython window and do:

`fig = plt.gcf()`
`ax = plt.gca()`
`ax.legend()`
`fig.canvas.draw()`

### Steps to reproduce the behavior
(1) Load some data
(2) Plot a spectra
(3) Add errors
(4) An extra none entry will appear in the legend 

### Platforms affected
Appears to be limited to workbench on Ubuntu.

DanielMurphy22, 2020-02-25:
Think its creating a Legend Title with the word `None` when it should be '' and so not displayed

DavidFair, 2020-02-25:
Quick speculation, could be an erroneous str cast like so

```
a = None
a = str(a)
plot(..., a)
```

PR #28147 (2020-03-02): Remove extra none in the legend of 1D plots
**Description of work.**
If the legend of a 1D plot was modified (by re-plotting data with errors for example) on Ubuntu an extra 'none' entry appeared in the legend. This was due to older versions of Matplotlib casting None to a string, see the commit https://github.com/matplotlib/matplotlib/commit/84292c46902913273d9231414e19bd11664117a6

This PR adds in a check before the legend is generated, to remove the 'None' string.

**To test:**

1. Open workbench on Ubuntu ( but presumably other distributions (and possibly mac?)
2. Plot any data
3. Right-cl [... cut]
closed 2020-03-02 by martyngigg