mantidproject/mantid #28104: Plot resets when moving start/end x markers in indirect data analysis

alicerussell1, 2020-02-26:
### Expected behavior
The changes should stay set

### Actual behavior
The plot goes back to the default zoom and scale

### Steps to reproduce the behavior
Open the indirect data analysis interface.
Go to one of the fitting tabs
Load a workspace
Right click on the plot and change the x axis scale to `log`
Move the marker
The plot resets

Zoom in on the plot
Move the markers
Plot resets 

### Platforms affected
Tested on window, bug is also in 4.2

Harrietbrown, 2020-03-02:
This is related to a more broad issue related to the context menu not properly communicating with the gui. This isn't likly to be fixed in this release and I have descovered another issue that can cause a crash through manipulating the context menu, so I have made a PR to dissable the menu for now as Mathew says he has not seen it bieng used by the scientists.

Harrietbrown, 2020-10-21:
After reenableing the context menu I cannot recreate the original issue. I think It is safe to put the context menu back.

PR #29795 (2020-10-26): Reenable the miniplot context menues in indirect data analysis
**Description of work.**

This PR reenables the context menu in indirect data analysis that were disabled in PR #28169
The context menu was dissabled as a temporary fix for issue #28104 but on reenabling the initial issue cannot be replicated.

**To test:**

1. Open the indirect data analysis interface.
2. Go to one of the fitting tabs
3. Load a workspace
4. Right click on the plot and change the x axis scale to log
5. Move the marker
6. the plot should stay in log scale
7. try interacting with the context menu in different ways to [... cut]
closed 2020-10-26 by martyngigg