mantidproject/mantid #28110: Tile plot not fully recovered by project recovery

giovannidisiena, 2020-02-26:
### Expected behavior
Tile plots should be recovered by project recovery.

### Actual behavior
Only one plot on the tile plot is recovered, see below:
[print.pdf](https://github.com/mantidproject/mantid/files/4256121/print.pdf)

### Steps to reproduce the behavior
Follow 4: http://developer.mantidproject.org/Testing/ErrorReporter-ProjectRecovery/ProjectRecoveryTesting.html

### Platforms affected
Workbench on Ubuntu 18, possibly all.

martyngigg, 2020-02-26:
@giovannidisiena Do you know if this worked in 4.3?

giovannidisiena, 2020-02-27:
@martyngigg no, it doesn't appear to be a regression.

ConorMFinn, 2020-03-02:
The functionality for project recovery loading was designed to only support figures with a single axes. Supporting multiple axes would be a feature change so it may be better to fix this as a maintenance issue?

martyngigg, 2020-03-02:
@ConorMFinn  I think you're right at this stage we should keep changes to a minimum. By all means do the work now but it should probably go on master unless the changes required are trivial.

PR #28188 (2020-03-04): Project Recovery: Recover Tiled Plots Correctly
**Description of work.**
Added functionality to save tiled plots in project recovery. 

**To test:**
1. Open workbench
1. Go to the settings and change the time between project recovery checkpoints to something bearable. 
1. Load some workspaces and create some tiled and conventional plots. 
1. Make a note of the plots and their appearance
1. Wait for project recovery to save the project
2. Crash mantid (using the SegFault algorithm)
3. Relaunch mantid 
4. Load from project recovery
5. Check that the recovered plots match what should have been sa [... cut]
closed 2020-03-04 by martyngigg