mantidproject/mantid #28123: ConvFit fit issues

Mantid-Matthew, 2020-02-27:
There seems to be two cases  in which convfit has some issues.

- If you do a fit with a resolution file which has more than one spectra the correct spectra resolutions are not assigned to the correct spectra in all cases.
- If you so a multiple input convfit with certain data there seems to be a 0 index error. I am not sure what is going on here and need to investigate.

SpencerHowells, 2020-02-27:
@Mantid-Matthew Have you got the data where the first problem occurs?

Mantid-Matthew, 2020-02-28:
@SpencerHowells If you want to replicate the first problem load inna non single spectrum resolution file and then select to not fit the first couple of spectra. The first couple of entries from the resolution file will still be used.

PR #28145 (2020-02-28): Convfit using correct resolutions
**Description of work.**

On the ConvFit tab loaded resolution files can have either one entry or the same number of entries as the input data workspace. In the case where the resolution workspace had the same number of entries as the data workspace entry each entry uses the corrosponding one from the resolution workspace.

If you loaded a workspace with many entries and then chose to only fit a section of these however you still just used the first n entries from the resolution workspace.

**To test:**
I have some data from a scientist which shows this issue  [... cut]
closed 2020-02-28 by NickDraper