mantidproject/mantid #28143: Crash on loading file and running script

DanielMurphy22, 2020-02-28:
**Original reporter:** Tatiana

very similar to: refs #27594 

### Expected behavior
Should be able to run a script and load a file at the same time
### Actual behavior
Workbench and plot crash
### Steps to reproduce the behavior
run this script (should take around 13 seconds, increase the for loop if you need longer!)
while its running load a file such as HRP39182.RAW or GEM40979.raw (should be anything!)
(in MantidPlot you may need to get the load dialog ready before starting the script)
### Platforms affected
Workbench and Plot Nightly before 5.0 release 4.2.20200225.1733

NickDraper, 2020-03-03:
so running this script does not affect Loading
``` python

for i in range(1000):
    Pause(0.1)
```

NickDraper, 2020-03-03:
this is also fine, with loading mar11060
``` python 

for i in range(10):
    ws = Load("GEM40979.raw") 
    ws2 = Load("HRP39182.RAW")
```

NickDraper, 2020-03-03:
but I can get it to crash with the script in the description

NickDraper, 2020-03-03:
Ok, so this is due to the sheer number of algorithms being run in the script, which is 10*200*200.

If you increase algorithms.retained in the properties file from 50 to 500, the problem goes away ...

NickDraper, 2020-03-03:
This should be used as a test case for #28093

PR #28280 (2020-03-27): Remove completed Algorithms from AlgorithmManager managed algorithms
**Description of work.**

Removes any finished algorithms from the AlgorithmManager list on the creation of a new algorithm.

**To test:**
1. Start several tabs
1. in each run some code a bit like this
   ``` python
   for i in range(600):
      Pause(0.1)
   ```
1. while running the tabs, run an algorithm from the Algorithm toolbox as well
1. Run StartLiveData as well
1. Just get lots of things running at the same time ...  We are testing we do not have any multithreading issues
1. #28143 has a good test case of a script tha [... cut]
closed 2020-03-27 by martyngigg