mantidproject/mantid #28146: Check if python algorithms CalculateMonteCarloAbsorption and SimpleShapeMonteCarloAbsorption need updating

DannyHindson, 2020-02-28:
The C++ MonteCarloAbsorption algorithm was recently updated to not use the NumberOfWavelengthPoints parameter by default. Check if the dependent Python algorithms CalculateMonteCarloAbsorption and SimpleShapeMonteCarloAbsorption should be updated to make it clear that their NumberOfWavelengthPoints parameter is no longer used (or possibly even remove it completely)

DannyHindson, 2020-05-20:
There is a PR in progress at the moment (#28462) to reorganize the code in these two algorithms. It looks likely that the two algorithms will live on but will be labelled deprecated.

To fully address this issue the NumberOfWavelengthPoints parameters should be removed from both algorithms since setting it has no effect. Also the new algorithm PaalmanPingsMonteCarloAbsorption should also not use this parameter. It's reasonable for all three to use ResimulateTracksForDifferentWavelengths=False which is the default.

SpencerHowells, 2020-05-20:
@DannyHindson I do not understand this. We could have 2000 wavelength points so we calculate for default 10 wavelengths to reduce computing time. This is acceptable because there is only a small wavelength variation & when applying the correct we do an interpolation. We have been doing this for 10 years!

DannyHindson, 2020-05-21:
@SpencerHowells - perhaps we can have a chat about this sometime? I can call you to discuss if you let me know when you're free? I'll try and answer here though in the meantime.

The MonteCarloAbsorption algorithm has two modes of operation which are driven by the boolean ResimulateTracksForEachWavelength parameter:

a) its default mode of operation involves generating one set of neutron tracks and then reusing them to calculate the attenuation factor for all wavelengths. The cost of the computation outside the track generation is v small so in this mode of operation it's fine to perform the attenuation factor calculation for all wavelengths in the input workspace without doing any interpolation - even if there are hundreds of wavelength points.

b) its other mode of operation is to pick a small set of wavelength points (eg 10) and generate a new set of tracks for each of these points + then do the attenuation factor calculation for these 10 points. When that's complete, the algorithm interpolates between the 10 points to generate the attenuation factor at all wavelengths. This is more like what you're describing. However since there aren't any wavelength dependent effects in the track generation there's no upside to doing this compared to approach a). So MonteCarloAbsorption uses the approach a) by default but gives the user the option to do the second approach if they prefer.

My most recent comment on this ticket relates to the fact that the default mode of operation was switched from b) to a) a couple of months ago. Some of the wrapper algorithms that sit around MonteCarloAbsorption still have the NumberOfWavelengthPoints parameter even though this isn't used in approach a).

SpencerHowells, 2020-05-21:
@DannyHindson Ok. I think that makes sense.

DannyHindson, 2021-06-28:
Closing because both algorithms have been labelled deprecated so not a priority

closed 2021-06-28 by DannyHindson