mantidproject/mantid #28149: Testing the exercises in the mantid basics tutorial for the training

RichardWaiteSTFC, 2020-02-28:
The training course has exercises for each of the sections below that need to be tested on workbench.

- [ ] Getting Started
- [ ] Loading and Displaying Data
- [ ] Algorithms, Workspaces and Histories
- [ ] Connecting Data To Instruments
- [ ] Fitting Data
- [ ] Mantid Plot
- [ ] Useful Extras

Build a docs-html target of Mantid and then comment on this issue which section you are testing so that other people will know. Tick the checkbox when complete and without issue.

RichardWaiteSTFC, 2020-02-28:
I will test:
Loading and displaying data
Mantid Plot

RichardWaiteSTFC, 2020-02-28:
In Algorithms, Workspaces and histories Ex1 - the last instruction should be point 7, and it's not clear what is required (just to show the history?)

PhilColebrooke, 2020-03-03:
I've found what looks like a formatting error on the Workbench Main Window Page
https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/getting_started/main_window.html
[image]

DavidFair, 2020-03-04:
I'll take algs, ws and histories

PhilColebrooke, 2020-03-04:
Do your APR David

DavidFair, 2020-03-04:
Some minor nit picks:
In "What is a Bin?...", 
- Typo `deteremines` and the comma afterwards is incorrect
- The colour fill plot has a completely different scale (which looks wrong), it looks the same in 4.2, so I don't think it's a regression for this release?
- `except for data collected in monitors, are as Event data.` -> `except for data collected in monitors, is in Event mode.`
- `Observe that as the bins get larger`, we should state something like `Observe in a plot for each that as the bins get larger`

In Algorithm histories:
- `paste what was copied to the clipboard.` You need to add, `without deleting the template code` since I instinctively did a CTRL+A / Del 

Interfaces:
- `and browse to the “loqdemo” folder` this is in ISIS Sample Data not the Training course data, it might be worth adding words to that effect

> Do your APR David

Already done :)

DanielMurphy22, 2020-03-05:
@DavidFair the Loqdemo folder is now in TrainingCourseData if you redownload it! Did the ISIS SANS interface example work for you?

DavidFair, 2020-03-05:
Yes, but I used my copy from the SampleData folder I had locally, rather than redownload hence I didn't notice the issue

DanielMurphy22, 2020-03-05:
Cool I'll fix

Harrietbrown, 2020-03-09:
Fixed by #28246

PR #28246 (2020-03-09): Fixes to MBC 
**Description of work.**
Many small fixes to mistakes in the MBC

**To test:**
Build docs if you wish.

Fixes #28149 

*This does not require release notes* because **documentation**

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code conform to the [coding standards](http://developer.mantidproject.org/Standards/)?
- Are the unit tests small and test the class in isolation?
- If there are changes in the releas [... cut]
closed 2020-03-09 by Harrietbrown