mantidproject/mantid #28155: Muon loader doesn't always load logs

AnthonyLim23, 2020-02-28:
Try to load MUSR 91350 and the Steer_... logs will be missing. The error:

```
Error in execution of algorithm LoadMuonLog:
Property with given name already exists search object SAMPLE_TEMP
Unable to successfully run LoadMuonLog Child Algorithm
Unable to successfully run LoadMuonLog Child Algorithm
```

is produced

NickDraper, 2020-03-02:
The first step here is to work out what is happening and is this something special about just this file.

What to do as a solution can only be decided after that.

AnthonyLim23, 2020-03-02:
I have asked for a file, the one above does not work. I did see the error the other day. It seems the alg just dies if it gets unexpected information. Leaving a half populated logs for a workspace

AnthonyLim23, 2020-03-03:
I had a look into this. The run is MUSR 81380. It does have two different entries from `Sample_Temp`. The first is loaded outside of this algorithm and is a single value (I assume the mean). The second is loaded as part of this algorithm and is a time series. 

Looking at other runs that do not fail suggest that the correct one is the single value.

PR #28241 (2020-03-10): LoadMuonLogs keeps processing when duplicate is found
**Description of work.**
LoadMuonLogs would stop loading logs once it found a duplicate. This PR fixes that.

**To test:**
Load MUSR 62260 - look at the sample logs. This is one the worked originally. 
Load MUSR 81380 - it has sample_temp twice. It use to stop loading everything after sample_temp. You should now get an warning and then it carries on.

Fixes #28155. 

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the [... cut]
closed 2020-03-10 by NickDraper