mantidproject/mantid #28178: Unconnected SLOTS in indirect interface

DavidFair, 2020-03-03:
### Expected behavior
Signals and slot connections should match

### Actual behavior
The following warning is emitted when opening the interface
```
Object::connect: No such slot MantidQt::CustomInterfaces::IDA::JumpFit::updateAvailableFitTypes() in ../mantid/qt/scientific_interfaces/Indirect/JumpFit.cpp:64
Object::connect:  (sender name:   'cbParameterType')
Object::connect: No such slot MantidQt::CustomInterfaces::IDA::JumpFit::updateAvailableFitTypes() in ../mantid/qt/scientific_interfaces/Indirect/JumpFit.cpp:66
```

### Steps to reproduce the behavior
- Open the Indirect interface

Harrietbrown, 2020-03-04:
Which indirect interface is doing this?

DavidFair, 2020-03-04:
Data analysis

The warning doesn't appear in the log, but will appear in the console where Mantid is running

PR #28221 (2020-03-04): Disconnected signals in Indirect Data Analysis
**Description of work.**

When you launch MantidPlot from a terminal and then open the data Analysis interface there are some warnings about signals connecting to slots which don't exist in the terminal. This PR removes these connections. These connections should have been removed [here](https://github.com/mantidproject/mantid/commit/2f062c103783931a789bc56c55b4027c62ac2436#diff-e1e66e448261b72ba5afba459d79ae16) but were missed.

**To test:**

  * Check that the warnings no longer appear in the terminal.
  * Run through the testing instructions fr [... cut]
closed 2020-03-04 by martyngigg