mantidproject/mantid #28194: Workbench is a bit slow

AnthonyLim23, 2020-03-03:
This is initially just an investigation. 

Indirect has been slow.

James (moun) has also said workbench is slow:
```
Workbench is really slow with anything using the ADS. For example the Muon Analysis interface when loading more than one run at a time. (ALC is less affected) Also as an example my script “Create2DALCMap” (from the script muon/Create2DALCMap.py in the Repository). Run it with parameters:
                FirstFile=//hifi/data/hifi00166860.nxs
                NumberOfRuns=121
                StartTime=0.1
                EndTime=10
                NBunch=20
                LogSelection=Field_Main_Target
                OutputWorkspace=ChooseAWorkspaceName
On Plot it takes 37.42 seconds (my desktop PC, mostly dominated by loading the files). On Workbench it takes 4m 17s: nearly 7 times slower! Workbench with the Progress bar hidden: 4m 21.3s (the same within error).
```

DanielMurphy22, 2020-03-03:
Looks related to #26675

martyngigg, 2020-03-04:
This is odd. The progress reporter is written in C++ so I would expect that is okay. My first guess would be lots of back and forth calls to Python from the C++ ADS notfications but we probably need to strap a profiler around this.

martyngigg, 2020-03-04:
I'd suggest [cProfile](https://docs.python.org/3/library/profile.html#module-cProfile) in combination with [snakeviz](https://jiffyclub.github.io/snakeviz/)

gvardany, 2020-03-06:
I see the same thing. A reduction that takes 3 seconds in MantidPlot takes 1 minute in Workbench, this is worrisome.

AnthonyLim23, 2020-03-06:
what data was it and did you use an interface or was it a script/algorithm?

gvardany, 2020-03-06:
I tried with an interface, Indirect -> Reduction (ILL)

martyngigg, 2020-03-06:
I had a spare 30 minutes so I thought I'd see if I could reproduce this. A couple of complications with this first. We'll need to be careful when comparing things as various OS caches are going to be in the way.

For example, I ran James' Create2DALC algorithm example in Workbench first and got an execution time of ~1m 10s. I then ran it in MantidPlot and got ~47s. I then restarted Workbench and ran it again and got ~48s - so basically the same. I think the disk caches have cached a lot of data and after loading 1 file the first time they don't have to do much loading at all.

@gvardany I'd tried a very basic reduction on IN16B in E/I-FWS mode with ILL/IN16B/083072.nxs and it took about 2 seconds. I'm assuming your example was more complicated?

gvardany, 2020-03-09:
It was just with 10 files, so that is long enough to feel the difference. I can send you some over, so you can try?

AnthonyLim23, 2020-03-09:
I know the indirect group have complained about windows specifically.

gvardany, 2020-03-09:
I attach 20 files of E/I FWS for IN16B. That should be enough to strike the difference.
[in16b-data-scans.zip](https://github.com/mantidproject/mantid/files/4305347/in16b-data-scans.zip)

martyngigg, 2020-03-09:
@gvardany That shows the difference and the disk caches don't seem to make a difference here. Thanks. Unfortunately my original choice of using `cProfile` doesn't work with Python threads so this is going to be way harder to investigate.

gvardany, 2020-03-09:
I see, but can we rule out that it is not the progress bar?

martyngigg, 2020-03-09:
I don't think it's the progress bar. Can I check that

```python
from mantid.simpleapi import *

data_dir='/media/data1/external-data/ill/workbench-performance/in16b-data-scans/'
runs=",".join([data_dir + str(r) for r in range(261777,261797)])
IndirectILLReductionFWS(Run=runs, OutputWorkspace='sample_red')
```
will do the same as the interface?

If it does then this is very strange because the above script takes ~4s to run whereas the interface takes ~40s on my machine.

AnthonyLim23, 2020-03-09:
The complaints we have had have all been C++ interface. The current suspect is the plotting. @Mantid-Matthew can you think of any other potential problems?

martyngigg, 2020-03-09:
Stackoverflow pointed me at a different profiler, [yappi](https://pypi.org/project/yappi/), that seems to be able to profile multithreaded code.

@AnthonyLim23 The profile I now have would agree that the plotting functions are being called a huge number of times.

martyngigg, 2020-03-09:
I suspect we have the ADS observer wired up to replot every ADS update...

martyngigg, 2020-03-09:
Here's the [profile](https://github.com/mantidproject/mantid/files/4305678/workbench.yappi.callgrind.tar.gz). It's only Python so no C++ calls unfortunately.

martyngigg, 2020-03-09:
I can confirm that commenting out https://github.com/mantidproject/mantid/blob/master/qt/widgets/plotting/src/Mpl/PreviewPlot.cpp#L6 drops the reduction time from ~40s to ~7s for the interface so the plot calls for even the plots that are on the other tabs are causing the largest issue.

Perhaps we can disable the ADS watching during processing?

AnthonyLim23, 2020-03-09:
That might be the way to go

martyngigg, 2020-03-09:
@Mantid-Matthew Would you be able to look at how risky that would be to do for this release? I don't think it's the whole problem but a huge factor, certainly for the Indirect interface.

gvardany, 2020-03-09:
Yes, would be great to sort this out for the release, otherwise we have to advice not to use the interfaces on workbench in the release mail.

Mantid-Matthew, 2020-03-09:
I've opened #28290to try and fix this.

PR #28290 (2020-03-09): Speed up workbench preview plots
**Description of work.**

On workbench if you had any of the C++ interfaces open changes the ADS could take a long time. This was at least partially becuase preview plot was replotting far more than neccecary. This updates the ADSwatchers in preview plot so that it only replots if a relevant workspace changes.

**To test:**
See the discussion in #28194 for how to test this issue.

Fixes #28194  

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####
 [... cut]
closed 2020-03-10 by martyngigg