mantidproject/mantid #28200: Replace header guards with `#pragma once`.

Harrietbrown, 2020-03-04:
All cases of header guards `#ifndef` should be replaced with `#pragma once`.
This should be done in chunks for each folder in Framework to avoid endless merge conflicts.

Algorithms (Is most likely the largest chunk)
API
Beamline
Catalog
Crystal
CurveFitting
DataHandling
DataObjects
Examples (very small)
Geometry
HistogramData
ICat
Indexing
Kernel
LiveData
MDAlgorithms
MPIAlgorithms
Muon
Nexus
NexusGeometry
Parallel
PostInstall
Properties
PythonInterface
RemoteAlgorithms
RemoteJobManagers
ScriptRepository
SINQ
TestHelpers
Types
WorkflowAlgorithms

Once all these are done the final step would be a find replace of the whole project to find any headers that are useing header guards that are outside of Framework.

martyngigg, 2020-03-05:
As part of this work could we also make sure we update the [`class_maker`](https://github.com/mantidproject/mantid/blob/master/buildconfig/class_maker.py) tool?

martyngigg, 2020-03-09:
I also just thought that we should update the coding standards page, more specifically [this](http://developer.mantidproject.org/Standards/CPPStandards.html#id5) section. I think it just needs to say something like use `#pragma once` instead of `#ifndef` header guards.

Harrietbrown, 2020-03-09:
I'll add a line to the `Layout of Header File` section since it doesn't actualy say what sort of guard to use at all.

PR #28365 (2020-03-16): Replace final header guards with pragma once
This commit removes the final header guard that had been missed by the auto replacer.

Fixes #28200

**To test:**
Any issues in this PR should be caught by jenkins. As this is the final replacement run a check on all `.h`, `.tpp`, or `.cpp` files for any header guards 
```
#ifndef GUARD_ID
#define GUARD_ID

/. some code ./

#endif
```
Any remaining should be: 
 - required parts of setting up tests.
 - part of a exturnal library
 - autogenerated

*This does not require release notes* because **It is an internal change to the codebase.**

---

#### Re [... cut]
closed 2020-03-16 by NickDraper