mantidproject/mantid #28226: Colorfill Plot throws Exception changing to log scale

DavidFair, 2020-03-04:
### Expected behavior
- Should be able to change the colour scale to log
### Actual behavior
- Throws exception
```
  File "/opt/mantidnightly/lib/workbench/plotting/figureinteraction.py", line 743, in _change_colorbar_axes
    datafunctions.update_colorbar_scale(self.canvas.figure, image, scale_type, image.norm.vmin, image.norm.vmax)
  File "/opt/mantidnightly/lib/mantid/plots/datafunctions.py", line 1024, in update_colorbar_scale
    if locator.tick_values(vmin=vmin, vmax=vmax).size == 0:
  File "/usr/lib/python3/dist-packages/matplotlib/ticker.py", line 2123, in tick_values
    "Data has no positive values, and therefore can not be "
ValueError: Data has no positive values, and therefore can not be log-scaled.
```

### Steps to reproduce the behavior
Run the following:
```
Load(Filename='GEM38370_Focussed.nxs', OutputWorkspace='GEM38370_Focussed')
ConvertUnits(InputWorkspace='GEM38370_Focussed', OutputWorkspace='ws_lambda', Target='Wavelength')
```
- Plot -> Colorfill
- Right click figure -> Colour Bar -> Log -> Stops working

PhilColebrooke, 2020-03-04:
So this issue happens when you try to change the colorbar scale to log and the scale's range contains negative values.

There's already a check for if the min value is zero so the simple way to fix it is to also check if the min value is less than zero, and adjust it accordingly.

However when looking into this I found that the Images tab in the figure options restricts the input of the min and max value so you can't make them less than zero, which raises a question of whether negative values should be allowed on the colorbar scale. Either way what's allowed in the Images tab and what's allowed in the dialog when you double-click the colorbar axis should match.

Any ideas @DanielMurphy22?

DavidFair, 2020-03-04:
@PhilColebrooke Out of curiosity how did Plot handle this?

PhilColebrooke, 2020-03-04:
I forgot Plot existed and didn't think to check what it does 😅 

When the scale is set to linear, any range is permitted.
When the scale is set to log and the min value is zero or negative, it is changed to slightly-above-zero.
When the scale is set to log and both the min and max values are negative, the range is changed to 1 to 1000 (?)

I think making Workbench match this behaviour is too big a task for this release since it's dealing with two different menus. Should a temporary fix be put in place in the meantime or should fixing this issue be postponed?

Harrietbrown, 2020-03-09:
Fixed by #28242

PR #28242 (2020-03-09): Fix unhandled exception when a colorfill plot's scale is set to logarithmic and negative limits are entered
**To test:**
Load a workspace, right-click it and select Plot->Colorfill.
Open figure options, go to Images tab.
Check that when the scale is set to Linear you can enter negative numbers for min value and max value.
Check that when the scale is set to Logarithmic any negative values become zero and you can't enter negative values.
Apply the settings with the scale set to Logarithmic and close figure options.
Double-click the colorbar axis, check that no exception is thrown if you enter a [... cut]
closed 2020-03-09 by Harrietbrown