mantidproject/mantid #28263: use cmake variables to set paths for VS instead of visual-studio.bat

DannyHindson, 2020-03-06:
The Mantid C++ source code currently needs to be opened in Visual Studio by running the build\visual-studio.bat file which sets up some environment variables via thirdpartypaths.bat and opens up Mantid.sln. If the solution is opened without using the visual-studio.bat file then it's not possible to run\debug MantidPlot.exe or the unit tests - they fail with errors relating to Boost or Poco not being found. There are also a few Qt related compilation errors where the rcc or uic utilities can't find zlib.dll (MantidQtIconsQt4, MantidQtWidgetsSliceViewerQt4, MantidQtWidgetsPlottingQt4)

I think we could avoid needing the visual-studio.bat file by using these cmake variables to set up the paths in the VS project files themselves:

- VS_DEBUGGER_ENVIRONMENT which populates a VS setting called "Environment" which can hold a list of environment variables to be set up when running a project. This setting can be seen on the project properties in Visual Studio and we'd use it to add to PATH here
- CMAKE_MSVCIDE_RUN_PATH which allows extra items to be added to PATH when running builds in Visual Studio

Both variables have been available since cmake 3.13

martyngigg, 2020-08-10:
I had to fiddle with the `VS_DEBUGGER_ENVIRONMENT` settings in #29088 so I added the `CMAKE_MSVCIDE_RUN_PATH` settings at the same time.

martyngigg, 2020-08-27:
This didn't turn out so easy as a single line setting.

closed 2021-02-23 by gvardany