mantidproject/mantid #28291: Add mechanism to MonteCarloAbsorption to restrict generation of scatter points to sample only

DannyHindson, 2020-03-09:
Request from Chris Ridley. The legacy absorption correction code they have for Pearl only generates scatter points in the sample and not the environment. So to support cross checking the Mantid results against the results from the legacy code it would help if Mantid's calculation could do the same.

Possible approaches:
- add explicit "Sample only" switch into MonteCarloAbsorption algorithm parameters
- add similar switch but attach it to the material
- allow the scattering cross section to be set to zero in a material + only generate scatter points if the cross section is greater than zero

SpencerHowells, 2020-03-09:
@DannyHindson @martyngigg I don't get this. If the environment is defined as just a Sample it can only scatter from the sample. I've been doing this for years.

DannyHindson, 2020-03-09:
Chris wanted to retain the attenuation effect of the surrounding environment components in the absorption correction without them generating any scatter points. If I dropped the environment components completely then I wouldn't get this behaviour.

DannyHindson, 2020-04-22:
Likely to be covered by the work on Paalman Pings

closed 2020-04-22 by DannyHindson

DannyHindson, 2020-06-16:
Closing because feature added as part of PR #28462