mantidproject/mantid #28306: Harmonise Window behaviour

DanielMurphy22, 2020-03-10:
After #27883, it was noticed that Windows behave somewhat differently across different OSs. 

This issue is to try to harmonise the behaviour of all pop-up windows in Mantid Workbench to the intended behaviour (what this is should be discussed).

The difficulty here is that different platforms use different window mangers, which treat separate windows differently.

At the moment, the fix in #28222, results in Plots being on top of everything (instrument viewer, settings) on Mac, and Linux windows not being minimisable.

DanielMurphy22, 2020-03-13:
An odd current behaviour, is that on Mac, while plots are currently not maximisable after the #28222 fix, if you produce a plot on Mac WB and click the generate a script button, and choose script to clipboard, then the maximise green button appears in the corner of the figure! It can be disabled again by clicking "Fit"

DanielMurphy22, 2020-08-04:
One option could be for all the interfaces to be always on top, a bit like plots, as has been implemented for the EngDiff GUI:  #29071

DanielMurphy22, 2020-09-22:
Jos mentioned on the [Forum](https://forum.mantidproject.org/t/interfaces-hide-behind-workbench-window/690) that plotting from the reflectometry gui hides the interface behind the main window. Foxing this issue should also solve this!
[image]

martyngigg, 2020-11-23:
For release 6 sprint 3 the focus is purely on improving the behaviour of the ISIS Reflectometry interface w.r.t plotting. Wider improevments will wait for the interface reorganization and dropping MantidPlot.

PR #30129 (2021-03-22): Add window setting (Floating/ontop) to workbench
**Description of work.**

This PR adds a setting to workbench to allow the floating and ontop behaviour of new windows to be toggled. Currently, this behaviour is limited to new plots, but could be extended to other windows, such as interfaces.

After a lot of trial-and-error, it was realised that a compromise would have to be reached with the settings. Ideally, we would get the same behaviour as on Windows, where ontop means:

- The window stays ontop of the main workbench window
- We have the common window functionality of minimize and maximiz [... cut]
closed 2021-03-22 by gvardany