mantidproject/mantid #28312: Unreliable Unit Test - RunsPresenterTest

giovannidisiena, 2020-03-10:
### Expected behavior
This test has been observed to fail by a number of developers - see https://builds.mantidproject.org/job/pull_requests-ubuntu/35472/ as an example.

### Actual behavior
It should pass!

### Steps to reproduce the behavior
Run the test.

### Platforms affected
Ubuntu, possibly all

giovannidisiena, 2020-03-11:
I have been running this on both Windows and Ubuntu: `for run in {1..N}; do ctest -C Debug -VV -R MantidQtScientificInterfacesTestQt4_RunsPresenterTest testStartMonitorUpdatesView | grep "1 tests failed"; done` however I am yet to see a failure after around 1000 runs.

martyngigg, 2020-09-04:
As this is related to a Qt4 test that will be going away with MantidPlot I'm going to close this one.

closed 2020-09-04 by martyngigg