mantidproject/mantid #28313: Clean/Dirty Install Smoke Tests

DanielMurphy22, 2020-03-10:
Before testing, Edit the Smoke Testing Umbrella Issue to add your name beside the OS you are running this test on.
Then tick the adjacent tick box once testing is complete.

Please run these tests on the compiled package of Mantid; **not a locally compiled version**.
Test MantidPlot AND Workbench!

Comment below with the OS you tested on
and any issues you came across.

If you have any questions please contact the creator of this issue.

## Checklist/directions

### Dirty install

* Make sure that you have several versions of Mantid installed
 * Last release
 * A nightly
 * If possible an old release
* Install the latest version of the new Mantid
- [ ] Check that Mantid boots up correctly

### Clean install

* Remove all existing Mantid versions and associated files
* Windows:
 * Uninstall the program
 * Clear shortcuts from desktop
 * Clean out the registry
  * Load regedit (Command Prompt > regedit)
  * Find `HKEY_CURRENT_USER > SOFTWARE > Mantid` delete it
* OSX:
 * Remove the application
 * Remove the `~/.mantid directory`
 * Remove `~/Library/Preferences/org.mantidproject.MantidPlot.plist`
* Linux
 * Remove the package: `/opt/Mantid`
 * Remove `~/.config/Mantid`
 * Remove `~/.mantid/`

* Re-install the latest version of the new Mantid
- [ ] Check that Mantid boots up correctly

ConorMFinn, 2020-03-13:
### Dirty Install
- Installing the `.deb` using the software installer did not overwrite the old version (4.2) of Mantid plot or workbench. 
- Installing via apt worked correctly.
### Clean Install
- Installed correctly using the Software Installer and apt. 

___
Tested on Ubuntu 18.04 | Release 5.0 Candidate (13-Mar-2020)

DanielMurphy22, 2020-03-13:
Worked fine for me, launched correctly.
Tested on Windows 10 | Release 5.0 Candidate (13-Mar-2020)
As many people as possible should test this on Windows

AnthonyLim23, 2020-03-13:
Used current nightly on my old desktop

Dirty install: I overwrote an old version of Mantid - everything seems fine

clean install: I uninstalled all version of Mantid at the same time. Installed nightly - everything opens ok

DanielMurphy22, 2020-03-13:
Worked fine
Tested on MacOS Catalina | Release 5.0 Candidate (13-Mar-2020)

ConorMFinn, 2020-03-13:
Tested on Windows 10 | Release 5.0 Candidate (13-Mar-2020)
___
### Dirty Install 
- Installation was paused due to workbench running while installing, but still completed. Could not open workbench, plot opened with a warning that python did not initialize. Any python interfaces, algoorithms, etc. could not be run. Attempting to install again failed. 

### Clean Install
- Required the manual deleting of the MantidInstall directory due to previous issues. 
- Otherwise installed and launched correctly.

DannyHindson, 2020-03-13:
Tested on Windows 10:

Dirty and clean install both worked for me. Two minor issues:

a) the uninstall leaves behind these files:
bin\debug.log
bin\Mantid.user.properties
lib\qt5\bin\debug.log

b) it took me a while to notice that the start menu entries are now "Mantid Plot" and "Mantid Workbench" with a space in them. I seem to remember the previous names were without the space so I kept failing to find the 5.0 install when typing MantidP… or MantidW...

closed 2020-03-16 by NickDraper