mantidproject/mantid #28314: Fitting Smoke Tests

DanielMurphy22, 2020-03-10:
Before testing, Edit the Smoke Testing Umbrella Issue to add your name beside the OS you are running this test on.
Then tick the adjacent tick box once testing is complete.

Please run these tests on the compiled package of Mantid; **not a locally compiled version**.
Test MantidPlot AND Workbench!

Comment below with the OS you tested on
and any issues you came across.

If you have any questions please contact the creator of this issue.

## Checklist/directions

* Fitting to data [directions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/01_fitting_models_to_data.html#fitting-models-to-data)
- [ ] Simple fit 
- [ ] Activating the fitting tools 
- [ ] Selecting the fitting range 
- [ ] Other Fit Function Settings properties 
- [ ] Setting up a fit model 
- [ ] Adjusting fit function parameter 
- [ ] Saving a model 
- [ ] Tying and constraining fit parameters 
- [ ] Execute your fit 
- [ ] Inspecting a fit result 
* Additional options [directions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/02_fit_model_choices.html#fit-model-choices)
- [ ] Non peak model + background 
- [ ] User defined function

DannyHindson, 2020-03-12:
Tested on Ubuntu in MantidPlot and Workbench using nightly build (4.2.20200311.1916). Few minor strange features in MantidPlot (eg Fit window still shows chi-sq of old fit in the window header even after having cleared model, fitted curves don't show up on plot if I open Fit Function from View menu) but these were present in 4.2. So overall looks fine

DannyHindson, 2020-03-13:
Repeated smoke test on release candidate (5.0) and results same as yesterday ie fine

DanielMurphy22, 2020-03-16:
had a brief look and seemed fine on MacOS Catalina 14/03/20

closed 2020-03-16 by NickDraper