mantidproject/mantid #28315: Interface Smoke Tests

DanielMurphy22, 2020-03-10:
Before testing, Edit the Smoke Testing Umbrella Issue to add your name beside the OS you are running this test on.
Then tick the adjacent tick box once testing is complete.

Please run these tests on the compiled package of Mantid; **not a locally compiled version**.
Test MantidPlot AND Workbench!

Comment below with the OS you tested on
and any issues you came across.

If you have any questions please contact the creator of this issue.

## Checklist/directions

- [ ] Mantid opens without errors or warnings 
- [ ] Every option in `Interface` opens a GUI 
- [ ] Load some test data 
- [ ] Catalogue opens  [directions here](https://www.mantidproject.org/Catalog_Search) 
- [ ] Saving/loading projects works  [directions here](https://www.mantidproject.org/MantidPlot:_The_File_Menu#File-.3E_Save_Project_as...)

ConorMFinn, 2020-03-11:
- Frequency Domain Analysis not opening in plot with the following error:
 ```
AttributeError: 'FrequencyAnalysisGui' object has no attribute 'resizeDocks'
  at line 29 in '<Interface>'
  caused by line 100 in '/opt/mantidnightly/scripts/Muon/GUI/FrequencyDomainAnalysis/frequency_domain_analysis_2.py'
```
- Muon Analysis also not opening with a similar error:
```
AttributeError: 'MuonAnalysisGui' object has no attribute 'resizeDocks'
  at line 29 in '<Interface>'
  caused by line 95 in '/opt/mantidnightly/scripts/Muon/GUI/MuonAnalysis/muon_analysis_2.py'
```
_____

Tested using Mantid Nightly 4.2.20200310 on Ubuntu 18.04

alicerussell1, 2020-03-12:
Testing on windows 10:
No issues found (the issue above has been fixed)

ConorMFinn, 2020-03-13:
Tested on Ubuntu 18.04 | Release 5.0 Candidate (13-Mar-2020)
___
- Tiled plots not recovered by loading a project in workbench, but this was also the case in 4.2.
- Interfaces -> Indirect -> Settings does not open in plot if another Indirect GUI is open, but this is also the case in 4.2.

DanielMurphy22, 2020-03-13:
All interfaces seem fine
Tested on MacOS Catalina | Release 5.0 Candidate (13-Mar-2020)

closed 2020-03-16 by NickDraper