mantidproject/mantid #28316: Plotting Smoke Tests

DanielMurphy22, 2020-03-10:
Before testing, Edit the Smoke Testing Umbrella Issue to add your name beside the OS you are running this test on.
Then tick the adjacent tick box once testing is complete.

Please run these tests on the compiled package of Mantid; **not a locally compiled version**.
Test MantidPlot AND Workbench!

Comment below with the OS you tested on
and any issues you came across.

If you have any questions please contact the creator of this issue.

## Checklist/directions

* 1D plotting:[instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/03_displaying_1D_data.html#displaying-1d-data)
 - [ ] Simple plot 
 - [ ] Another way to plot 
 - [ ] Adding curves to existing plots 
* 2D plotting: [instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/04_displaying_2D_data.html#displaying-2d-data)
 - [ ] Plot all spectra 
 - [ ] Plot from matrix 
 - [ ] Change colour map 
 - [ ] Contour plot 
 - [ ] Spectrum viewer 
* 3D plotting:
 - Load some data eg `LOQ7041` from the ISIS sample data
 - Double click the workspace to open the table view
 - Select the table
 - [ ] 3D plot becomes available as an option in the toolbar 
 - [ ] 3D wire frame 
 - [ ] 3D hidden line 
 - [ ] 3D polygon 
 - [ ] 3D wire surface

PhilColebrooke, 2020-03-11:
I've found that on Workbench if you change the axis limits on a plot, then changing either or both axis scale(s) to Log (via the right-click menu) makes the plot go wacky:
[image]

To reproduce:
1. Make a plot.
2. Change either or both axis limits, such as by zooming/panning.
3. Right-click the figure and select Axes -> any option except Lin x/Lin y

PhilColebrooke, 2020-03-12:
Made an issue for the above #28358

giovannidisiena, 2020-03-12:
Opened #28364.

DanielMurphy22, 2020-03-12:
Tested on MacOS Catalina, all fine.

giovannidisiena, 2020-03-13:
No new issues on Ubuntu.

DanielMurphy22, 2020-03-13:
All seems fine
Tested MacOS Catalina | Release Candidate 5.0 13/03/2020

PhilColebrooke, 2020-03-16:
One thing I've found that I'm gonna assume isn't a big deal:
If you do a colorfill plot of INTER00013464, right-click it and change the colorbar scale to Log, and then right-click it again and change the normalisation to None, you get `Warning: converting a masked element to nan.` show in the message log over and over. The warning keeps appearing anytime you move the cursor over the plot. This wasn't in the last release as changing the normalisation on 2D plots is new to this release, and I don't know if the file I used is the only one it happens on.

closed 2020-03-16 by NickDraper