mantidproject/mantid #28317: Python Smoke Tests

DanielMurphy22, 2020-03-10:
Before testing, Edit the Smoke Testing Umbrella Issue to add your name beside the OS you are running this test on.
Then tick the adjacent tick box once testing is complete.

Please run these tests on the compiled package of Mantid; **not a locally compiled version**.
Test MantidPlot AND Workbench!

Comment below with the OS you tested on
and any issues you came across.

If you have any questions please contact the creator of this issue.

## Checklist/directions

* Test that the Python scripting window works, [directions here](http://docs.mantidproject.org/v3.7.2/interfaces/ScriptingWindow.html)
 - [ ] Editor options 
 - [ ] Execution options 
 - [ ] Script output  
- [ ] Perform some workspace algebra 
- [ ] Test numpy functionality 
- [ ] Use the scripting window to run some scripts 
- [ ] Run through some examples from [the documentation](http://www.mantidproject.org/Python_In_Mantid), 3 or 4 examples is enough

ConorMFinn, 2020-03-11:
All seems to work fine. No major issues found. 
___
Tested on Ubuntu 18.04

giovannidisiena, 2020-03-12:
All looks good on Windows too 👍

giovannidisiena, 2020-03-13:
As above.

DanielMurphy22, 2020-03-13:
All seems fine on MacOS Catalina | Release Candidate 5.0 13/03/2020

ConorMFinn, 2020-03-13:
Tested on Ubuntu 18.04 | Release 5.0 Candidate (13-Mar-2020)
_____

Seems to be working fine.

closed 2020-03-16 by NickDraper