mantidproject/mantid #28318: Workspace/Algorithm Smoke Tests

DanielMurphy22, 2020-03-10:
Before testing, Edit the Smoke Testing Umbrella Issue to add your name beside the OS you are running this test on.
Then tick the adjacent tick box once testing is complete.

Please run these tests on the compiled package of Mantid; **not a locally compiled version**.
Test MantidPlot AND Workbench!

Comment below with the OS you tested on
and any issues you came across.

If you have any questions please contact the creator of this issue.

## Checklist/directions

* Load some workspaces, try 5 from different instruments
* Show detectors, check detector view GUI, follow non-Python directions [in the documentation](https://www.mantidproject.org/MantidPlot:_Instrument_View)
  - [ ] Render tab 
  - [ ] Draw tab 
  - [ ] Pick tab 
  - [ ] Instrument tree tab 
* Test 5 - 10 random algorithms, follow [the documentation](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/algorithms_workspaces_and_histories/02_algorithms.html#algorithms)
 - [ ] Help button 
 - [ ] Validation of inputs 
 - [ ] Overwriting workspace 
 - [ ] Run the algorithm 
- [ ] Check right-click options of the workspaces, they are listed [in the documentation](https://www.mantidproject.org/MantidPlot:_The_Workspace_Menu#Save_Nexus).

PhilColebrooke, 2020-03-12:
All good on Windows 10 I think

ConorMFinn, 2020-03-13:
Tested on Ubuntu 18.04 | Release 5.0 Candidate (13-Mar-2020)
_____
All seems fine, both on plot & workbench

gvardany, 2020-03-16:
Looks good on macOS, both plot & workbench.

closed 2020-03-16 by NickDraper