mantidproject/mantid #28319: Smoke Testing Umbrella Issue

DanielMurphy22, 2020-03-10:
This is to keep track of the coverage of smoke tests during the release cycle.
Edit this Issue and Mark beside your OS on the Test you are going to complete.

eg. Windows  - @DanielMurphy22

## Clean/Dirty Install Smoke Tests: #
 - [ ] Windows 
 - [ ] OSX  
 - [ ] Linux
 
## Fitting Smoke Tests: #
 - [ ] Windows 
 - [ ] OSX 
 - [ ] Linux
 - [ ] IDaaS  

## Interface Smoke Tests: #
 - [ ] Windows  
 - [ ] OSX 
 - [ ] Linux
 - [ ] IDaaS 

## Plotting Smoke Tests: #
 - [ ] Windows  
 - [ ] OSX 
 - [ ] Linux 
 - [ ] IDaaS  

## Python Smoke Tests: #
 - [ ] Windows  
 - [ ] OSX  
 - [ ] Linux  
 - [ ] IDaaS  

## Workspace/Algorithm Smoke Tests: #
 - [ ] Windows 
 - [ ] OSX 
 - [ ] Linux  
 - [ ] IDaaS 

 **Manually make this Umbrella into an Epic!**

gvardany, 2020-03-13:
I’m getting an Attribute Qt::AA_EnableHighDpiScaling must be set before QCoreApplication is created. message in terminal for workbench on macOS.

martyngigg, 2020-03-16:
We noticed this too but it's not currently doing any harm and most users won't see it. @DanielMurphy22 I think we created an issue for this didn't we?

DanielMurphy22, 2020-03-16:
Didn't create a separate issue for this, just mentioned it at the bottom of #27244 (In this case the user was running mantid from the command line.
We can make a separate issue for this.

closed 2020-03-16 by NickDraper