mantidproject/mantid #28343: Error bars on gaussian fits are unrealisticaly large despite a good fit.

Harrietbrown, 2020-03-11:
### Expected behavior
A gaussian fit that is a reasonable match to the data should not have error bars many times larger than the curve.

### Actual behavior
Performing a gaussian fit following mantid introduction tutorials results in a fit with massive errors
https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/02_complex_case.html
### Steps to reproduce the behavior
1. Open Workbench.
2. Load `HRP39182.raw`.
3. Plot the first spectra.
4. Room into the peak around 93000.
5. Open the fitting gui
6. Fit a gaussian and flat background.
7. Check the output workspace, the error on the gaussian is much larger that expected, especially around the peak.

### Platforms affected

RichardWaiteSTFC, 2020-09-07:
Had a quick look at this and think the bug is summarise in the following figures - the errorbars on the evaluated curve spectrum are anomalously large - I see the same thing for both possible  y-axis normalisation (none/bin width).  I think that the error on the fitted curve can actually be set to zero: it represents an model for the underlying probability distribution function and the meaningful errorbars are those on the measured data.  In this way when the user selects to show the errorbars the fitted curve won't have errorbars that overlap with the data and will make a much clearer figure. The uncertainty on the fitted curve comes from the confidence on the fitted parameters.
[image]

RichardWaiteSTFC, 2020-09-09:
The error on the fit curves are evaluated using the covariance matrix of the fit and a jacobian that hold the derivative of the fit value at each data point wrt the individual parameters. This is valid (see [1]) but there might be an bug. You would expect the error on the fit to be comparable (hopefully smaller for a decent model) on the individual data points - yet in this case they are ~10^4 larger. The relevant file is https://github.com/mantidproject/mantid/blob/release-next/Framework/CurveFitting/src/IMWDomainCreator.cpp#L408

[1] http://homepages.inf.ed.ac.uk/rbf/CVonline/LOCAL_COPIES/DAVIES1/rd_bhatt_cvonline/node9.html#SECTION00041000000000000000

martyngigg, 2020-09-16:
@RichardWaiteSTFC Is this something we intend to do for 5.1?

RichardWaiteSTFC, 2020-09-16:
I don't think so - at least I need to think about how to test this because I can't spot anything wrong with the formula used. I think it's going to be a case of simulating fake data and testing it that way. It's not a quick fix (if it's wrong at all) - and I think we said given no one has complained about it we can leave it for 6S1? if that's OK with you?

RichardWaiteSTFC, 2020-09-28:
There might also be an issue with how the errors on the individual functions are propagated? The error on the bg+peak curve doesn't look consistent with the errors on the curves of the individual bg and peak workspaces from a fit
[image]

DannyHindson, 2020-10-27:
@martyngigg - just doing our diffraction sprint planning and don't think we'll have capacity to pick this up in our next sprint or two so will drop the diffraction label. Are you able to schedule in the core team?

martyngigg, 2020-10-27:
@DannyHindson I can add it to our ice box. We're at least a bit further forwards with the investigations that Richard has done.

closed 2020-11-19 by martyngigg