mantidproject/mantid #28389: Sample scatter points based on depth into sample\environment components

DannyHindson, 2020-03-18:
The MCInteractionVolume class used in the MonteCarloAbsorption algorithm currently samples scatter points in the sample or each environment component evenly across the whole volume. More scatter points should be at the "front" of the volume compared to the back due to absorption

SpencerHowells, 2020-03-18:
@DannyHindson that is not we have been doing in the traditional method for 50 years. See the ATLAS or Gudrun manuals.

DannyHindson, 2020-07-08:
After further thought (and noting Spencer's comment) I'm closing this because don't think it would be correct to make this change. I think the inverse exponential attenuation factors that are calculated correctly take into account any "depth" effects.
Perhaps my thinking here was driven by an (incorrect) interpretation of the simulated scatter point and tracks as individual neutron tracks. I think it's better to think of the tracks as paths that a set of neutrons travel along and the exponential factor says how many of the "set" reach the detectors.

closed 2020-07-08 by DannyHindson