mantidproject/mantid #28424: Crash - close plot, run script declaring a function

DanielMurphy22, 2020-03-24:
**Original reporter:** James Lord / ISIS

(what I was calling JL_18032020 and JL_20032020)

Investigated with @alicerussell1 

Crash after closing plot and running script which declares a function

### Steps to reproduce the behavior
Plot MAR11060, spectrum 1.
Then close the plot.
Run this script (you may have to run it 2 or 3 times)
```
from mantid.api import *
import numpy

class Example1DFunction(IFunction1D):

    def init(self):
        self.declareParameter("C", 0.0)

    def function1D(self, xvals):
        # Access current values during the fit
        c = self.getParameterValue("C")

        return c*numpy.sqrt(xvals)

FunctionFactory.subscribe(Example1DFunction)
```
Below the script editor it will say "Running" and then mantid will crash
### Platforms affected
Workbench all, tested mostly on 5.0 on Windows 10

NickDraper, 2020-03-31:
So this appears all to be about notifications from the FuinctionFactory, all FitPropertyBrowsers (there's one for each plot in workbench) subscribe to an update notification.  But when the graph has been deleted the notification is still going to the now deleted object and causing a crash.

closed 2020-04-07 by martyngigg

DanielMurphy22, 2020-04-07:
I've emailed James now this has been resolved

PR #28465 (2020-04-07): Workbench crash after closing a plot then defining a new fit fn
**Description of work.**
This PR solves two issues.

1. It removes the observer from the FitFunctionBrowser on the FitFunctionFactory in its destructor.  This was what was causing the crash.
1. It also stops the FitFunctionBrowser being initialised twice, which confused the bejeebus out of me as to why number 1 did not work initially, as this meant there were two observers registered for each FitFunctionBrowser instance. 

**Report to:** @DanielMurphy22 -> James Lord

**To test:**
### Steps to reproduce the behavior
Plot MAR11060, [... cut]