mantidproject/mantid #28445: Workbench can't count the number of new script tabs

DavidFair, 2020-03-27:
### Expected behavior
Every new tab should go
New tab
New tab (1)
New tab (2)...etc.

### Actual behavior
We get
New tab
New tab
New tab (1)
New tab (1)
New tab (2)
New tab (2)

### Steps to reproduce the behavior
Open workbench and hit CTRL+N a lot

ConorMFinn, 2020-03-27:
I was unable to reproduce this on Ubuntu 18.04

DavidFair, 2020-03-30:
Kubuntu 18.04, so we should be seeing the same thing
This is on a clean build of master from today (2020-03-30)
[image]

martyngigg, 2020-03-30:
@DavidFair I am also unable to reproduce this on Ubuntu 18.04 with GNOME on the latest nightly from our apt repos.

martyngigg, 2020-03-30:
I don't quite understand why it would be Kubuntu-specific though...

DavidFair, 2020-03-30:
Maybe it's how I'm creating them, I'm holding CTRL and tapping N, without letting go of the CTRL key?

ConorMFinn, 2020-03-30:
I was doing the same and couldn't recreate it, does Kubuntu handle tabs in some subtly different way?

DavidFair, 2020-03-30:
Possibly, I guess as the reporter and only replicator it falls on me to investigate

martyngigg, 2020-03-30:
https://github.com/mantidproject/mantid/blob/master/qt/python/mantidqt/widgets/codeeditor/multifileinterpreter.py#L67 is where the counting happens. It's not immediately obvious why that's wrong...

PR #28452 (2020-03-30): Create new tab names correctly on KDE
**Description of work.**
Creates new tab names correctly on non-GNOME enrionvments for Ubuntu.
KDE automatically binds keyboard shortcuts, which Qt represents with &x
(where x is the shortcut). This breaks the name lookup as &New != New in
"New Tab".

We now strip out these characters when looking up the names fixing the
names not incrementing correctly.

**To test:**
- Check tab names still increment correctly
- Either take my word they are fixed, or install Kubuntu/KDE DE and try it out

Fixes #28445 . 
---

#### Reviewer ####

Please comment on the foll [... cut]
closed 2020-03-30 by martyngigg