mantidproject/mantid #28458: Also output Batch File name in CanSAS1D and NXcanSAS

smk78, 2020-03-31:
Currently SaveCanSAS1D and SaveNXCanSAS write the User File name to the SASprocess field in the metadata.

If the data being output was generated by processing a Batch File we should also write the Batch File name for enhanced traceability.

DavidFair, 2020-03-31:
Adding some info here so that other devs can pick this up too:

Looking at the source file `SaveCanSas1D2.cpp:215` we save the user file as a term in `SASprocess` whose spec is found here:
http://www.cansas.org/formats/canSAS1d/1.1/doc/element_SASprocess.html

We need to add a new field @smk78 do you have any preference on this XML tag naming since it's effectively free-form?

I'm thinking `batch_file` as follows:

```xml
<SASprocess>
    <term name="user_file">user_file_name.txt</term>
    <term name="batch_file">batch_file_name.csv</term>
```

I'm also not sure if the batch file is added to the logs so we'd need to do that too, but that should be relatively simple from the SANS side

smk78, 2020-03-31:
I'm happy with that XML tag naming!

smk78, 2020-09-18:
Looking at the testing I have performed over the last week or so using 5.0.20200909.1539 and 5.0.20200914.1537 I note that not one of the CanSAS1D xml files (I haven't checked the NXcanSAS) contains a batch file entry.

I am attempting to check with the latest nightly, but are we sure this issue was fixed?

gemmaguest, 2020-09-18:
It's there for me! But I'll double check on Windows.

[image]

gemmaguest, 2020-09-18:
This is fine for me in the latest Windows nightly build too - both CanSAS1D and NXcanSAS contain `user_file` and `batch_file` entries. I am using the training course loqdemo.

DavidFair, 2020-09-18:
I've just tested:

- Last nights nightly on Windows / Linux on two different machines
- A latest build of `release-next`

And it seems to be working?

@smk78 is it worth checking that the save is set to file or both and the files aren't existing ones from previous testing?
Otherwise I'm stumped.

smk78, 2020-09-18:
Well that's odd! @DanielMurphy22 pointed me at your Jenkins server for a 'current' nightly and when I do this
[image]
I get this
```
			<name>Mantid generated CanSAS1D XML</name>
			<date>18-Sep-2020 11:19:03</date>
			<term name="svn">5.0.20200917.1709</term>
			<term name="user_file"></term>
			<term name="batch_file"></term>
			<term name="sample_trans_run">108174</term>
			<term name="sample_direct_run">108173</term>
			<term name="can_trans_run">108175</term>
			<term name="can_scatter_run">108177</term>
			<term name="can_direct_run">108173</term>
```
and
```
			<name>Mantid generated CanSAS1D XML</name>
			<date>18-Sep-2020 11:19:03</date>
			<term name="svn">5.0.20200917.1709</term>
			<term name="user_file"></term>
			<term name="batch_file"></term>
			<term name="sample_trans_run">108174</term>
			<term name="sample_direct_run">108173</term>
			<term name="can_trans_run">108175</term>
			<term name="can_scatter_run">108177</term>
			<term name="can_direct_run">108173</term>
```
Here are the User and Batch file I was using
[startup202.zip](https://github.com/mantidproject/mantid/files/5244927/startup202.zip)

What is going on?!

smk78, 2020-09-18:
Also see https://github.com/mantidproject/mantid/issues/28847

smk78, 2020-09-18:
> @smk78 is it worth checking that the save is set to file or both and the files aren't existing ones from previous testing?
> Otherwise I'm stumped.

[image]

[image]

And the date/SVN in my output above shows you I'm looking at recently generated files!

smk78, 2020-09-18:
Oooo... this is interesting. You will notice that the reductions above were merged reductions. If I change the User File to one that only reduces the LAB, the User File appears in both the CanSAS1D output and the NXcanSAS!
```
			<name>Mantid generated CanSAS1D XML</name>
			<date>18-Sep-2020 11:51:06</date>
			<term name="svn">5.0.20200917.1709</term>
			<term name="user_file">USER_LOQ_202A_M3_Team_LOQ_8mm_Changer_MAIN_LIN.txt</term>
			<term name="batch_file"></term>
			<term name="sample_trans_run">108176</term>
			<term name="sample_direct_run">108173</term>
			<term name="can_trans_run">108173</term>
			<term name="can_scatter_run">108179</term>
			<term name="can_direct_run">108173</term>
```
But the batch file is still missing.

smk78, 2020-09-18:
I may have found something.

This latest nightly lets you export the table to a Mantid-generated batch file again. So I did that. Now when I reload that exported batch file it shows up in the files.
```
			<name>Mantid generated CanSAS1D XML</name>
			<date>18-Sep-2020 12:02:47</date>
			<term name="svn">5.0.20200917.1709</term>
			<term name="user_file">USER_LOQ_202A_M3_Team_LOQ_8mm_Changer_MAIN_LIN.txt</term>
			<term name="batch_file">exported_batch_file.csv</term>
			<term name="sample_trans_run">108176</term>
			<term name="sample_direct_run">108173</term>
			<term name="can_trans_run">108173</term>
			<term name="can_scatter_run">108179</term>
			<term name="can_direct_run">108173</term>
```
unless you do a merged reduction
```
			<name>Mantid generated CanSAS1D XML</name>
			<date>18-Sep-2020 12:06:26</date>
			<term name="svn">5.0.20200917.1709</term>
			<term name="user_file"></term>
			<term name="batch_file"></term>
			<term name="sample_trans_run">108176</term>
			<term name="sample_direct_run">108173</term>
			<term name="can_trans_run">108173</term>
			<term name="can_scatter_run">108179</term>
			<term name="can_direct_run">108173</term>
```
Arrgghhh.

smk78, 2020-09-18:
Here are the two Batch Files I was using (you have the two User Files in the earlier upload). Why are these causing different behaviour in the output fields when both load and Mantid will process both?
[startup202-b.zip](https://github.com/mantidproject/mantid/files/5245136/startup202-b.zip)

DavidFair, 2020-09-18:
So the difference is that your using a custom mask file - this is why the batch file and user file isn't being copied through (an easy fix for the same reason as the 2D button)

I'm just looking at how `merged` interacts....

closed 2020-09-22 by martyngigg

smk78, 2020-09-23:
Testing Nightly 5.0.20200922.1715 this seems to be working. Also tested the loading of the CanSAS1D (XML) and NXcanSAS (HDF5) output with this field in SasView 5.0.3. They load and display correctly.

PR #29543 (2020-09-22): Fix edge cases with user filename
**Description of work.**

Fixes the following cases where the user filename and batch filename were not output:
- Custom user file in the table
- Merged reductions

Additionally fixes constantly switching the instrument on table row change. This was because the file information equality operator was not implemented, causing us to reload file info every time

**Report to:** @smk78 

**To test:**
- Download the ISIS Sample Data -> LOQ demo
- Open the SANS GUI
- Load in Maskfile.txt to user file
- Load in the .csv from the LOQ Demo folder
- In the bottom right se [... cut]