mantidproject/mantid #28484: Indirect Data Analysis minor fixes

Mantid-Matthew, 2020-04-06:
The following have been identified as issues still remaining with the Indirect Data Analysis GUI.

MsdFit
====
msdFit - default values not taken from data
Msd, Beta & Sigma parameters need constraint >0
Change ws names to eg runname_msdFit_seqGauss_s0...

ConvFit
=====

Not all functions in QE folder
Check that WS number & Q change when spectrum changed
Lor doesn't get res from IPF
Carry over Lor1 paras when changing to 2 Lor?
Delta centre not displayed in small window
Delta height constraint >0
Check all functions for constraint
ws names to include seq/sim
Make number of Lor a spinner

FqFit
===
constraints on all
better default values?
WS is renamed _FWHM which is confusing

gemmaguest, 2020-04-21:
Re-opening because I believe the PR was only a partial fix

Harrietbrown, 2021-09-06:
Not stale

PR #28498 (2020-04-21): Correcting naming and constraints in Indirect Data Analysis
**Description of work.**

This solved some of the issues listed in #28484, more specifically it corrects the issues related to work space naming and constraints. 

Constraints have been added to all the tabs and the work space naming has been improved in the Msd and FQ fit tabs.

**To test:**
  * Data to test the relevant tabs can be found in #26631
  * Do a git on the MSD and FQ tabs and check that the workspace names contain information about what sort of fit was done and make sense. 
  * Check that all fit functions on all tabs hav [... cut]
closed 2022-12-20 by stale[bot]