mantidproject/mantid #28499: Slow launch Workbench

DanielMurphy22, 2020-04-09:
**Original reporter:** Rob Dalgliesh

Launching workbench has taken as long as 84 seconds.

When I tried on Windows, I got 24s to launch, but only 17s when `UpdateInstrumentDefinitions.OnStartup = 0` was added to the user properties file. But Loading IDFs shouldn't slow down launch time as it should be asynchronous.

It also seems that loading Python algorithms may be a major cause for any slowdown.

Method for capturing this slow-down has been to run [Process Monitor](https://docs.microsoft.com/en-us/sysinternals/downloads/procmon) with a filter added to only display python processes.

NickDraper, 2020-04-20:
Start by instrumenting and documenting the workbench startup times

DanielMurphy22, 2020-04-27:
This is likely due to having Network drives listed in Manage User Directories

DanielMurphy22, 2020-05-26:
#12425

DanielMurphy22, 2020-07-22:
@ConorMFinn Can this issue be closed? Or do we need it tested?

ConorMFinn, 2020-07-24:
We've made improvements to the problems mentioned in the issue, so I think we can now.

closed 2020-07-24 by ConorMFinn