mantidproject/mantid #28500: Plotting Suggestions

DanielMurphy22, 2020-04-09:
**Original reporter:** Rob Dalgliesh

It is worth deciding how viable and how much effort each of these features would be. Some of them could require very little effort.

Suggested improvements for Workbench plotting:
- [ ] MantidPlot add/remove curve menu
- [ ] A Copy toolbar button (currently you can save an image to file, but cannot copy and simply paste into a document). I suggest the current save button could have a drop-down (like generate a script) which lets you choose to Save or Copy (save to file or clipboard essentially!)
- [ ] Ability to add minor gridlines (this would tie in minor ticks)
- [ ] Able to set a default legend position (in Settings). This could be the [10 options in mpl](https://matplotlib.org/3.2.1/api/_as_gen/matplotlib.pyplot.legend.html).
- [ ] Legends Editable in-situ, like axis labels and titles are

NickDraper, 2020-04-21:
My comments:
- **Add/remove curve**
  So workbench does not work like mantidplot so the add from other graphs might be more difficult, but we could certainly give a way to remove the existing curves.  We already have a way to add new curves onto a graph using overplot.  Do we have any other ways to remove a curve from a plot?
- ** copy to clipboard for plot**
  Yes we certainly should allow this.  At least copying as an image.
-- ** Add minor gridlines "" -Yes
-- ** default legend location in settings ** - Yes
-- ** Legencds editable in situ ** - Depends on how easy this is to do with matplotlib

NickDraper, 2020-04-21:
Phil Colebrooke:house_with_garden:  1 hour ago
My thoughts on each suggestion:
1. I don't really see the benefit of this as you can already add curves via overplotting and remove curves in the figure options.
1. I was working with the toolbar recently so I saw that saving a figure is something built into matplotlib but it doesn't look like that's the case for copying to clipboard. I know on Windows you can press windows key + shift + S to screenshot a portion of your screen and it's copied to your clipboard, which does basically the same thing.
1. Easily doable I think.
1. I don't know too much about how the settings work but it seems like this should be doable.
1. Also seems doable, you can set curve labels in the figure options so it would just be putting that functionality into its own dialog which opens when you double-click on the legend.

DanielMurphy22, 2020-04-21:
I agree that the Add/remove menu is covered by Figure Options and Overplotting. I have created separate issues for each of the other points.

closed 2020-04-21 by DanielMurphy22