mantidproject/mantid #28529: Make Q values update correctly in FunctionBrowser

Mantid-Matthew, 2020-04-20:
Currently Q values do not update in the function browser when the dataset being observed is changed. This is due to a the FunctionBrowser not handling attritubute updating. This needs to be resolved so the Q values used are resolved correctly.

Harrietbrown, 2020-09-30:
an example of how to see where this is coming from:

go into conv fit and load data
fit with Tashirawater
show full function
see the Q value and workspace index in the fit browser is wildly off

StephenSmith25, 2020-09-30:
This should wait until I've made the changes to setting/getting attributes in composite function. Also, I've already made a start on this last sprint and got it so it shows the correct Q Values, so if you're picking this up I can show you where I got.

Harrietbrown, 2020-09-30:
Thanks, I am just going through issues to get a list of urgent things to fix before the maininance work.

PR #29810 (2020-11-11): Show correct attribute values in Indirect Data Analysis FunctionBrowser
**Description of work.**
This PR adds a missing functions to manipulate attributes in `IFunctionModel` and `IFunctionView`. Both classes were missing `getAttribute` and `setAttribute` methods. Due to changes made in a previous PR, these can now be dealt with in the exact same manner as `setParameter(..)`.

The primary motivation for this work was to show the correct `Q` value for `Q` dependent functions in the `ConvFit` tab of the indirect data analysis GUI, as such as the testing instructions will relate to this issue:

* [... cut]
closed 2020-11-11 by martyngigg