mantidproject/mantid #28530: Indirect fitting functions missing

Mantid-Matthew, 2020-04-20:
Some of the Indirect fitting functions are missing from the function browser. Investigate where they have gone

AnthonyLim23, 2020-08-03:
@Mantid-Matthew could we have some more detail please?

Mantid-Matthew, 2020-08-25:
There are some functions which are not appearing correctly in the GUI. Talk to Spencer Howells to find out which ones.

SpencerHowells, 2020-08-25:
@AnthonyLim23 These are complicated functions which Stephen looked at. Needs to be re-thought.

AnthonyLim23, 2021-01-15:
@SpencerHowells do we want to keep this open or do we know what to do next?

SpencerHowells, 2021-01-15:
@AnthonyLim23 Please keep open. I have some ideas to discuss.

AnthonyLim23, 2021-02-09:
@SpencerHowells could you arrange a time to discuss your ideas?

Harrietbrown, 2021-02-16:
The three missing functions are DiffSphere, DiffRotDiscreteCircle, IsoRotDif
These are imutable composit functions, that can not be edited, and therefore can not be added to the browser. A solution would be to make a new function that simply call this composit.

PR #30754 (2021-02-25): Add missing indirect fit functions
**Description of work.**
This PR adds a set of three fit functions that were not available previously in the fit browser (`IsoRotDiff`, `DiffSphere` and `DiffRotDiscreteCircle`). These are ImmutableCompositeFunctions that were not added.

**To test:**

1. Load some data.
2. Plot data
3. Open the fit browser.
4. Add `IsoRotDiff` and try to fit,
5. Repeat with `DiffSphere` and `DiffRotDiscreteCircle`
6. Open Indirect Data analysis
7. Go to the convfit tab
8. Select `DiffSphere` and `DiffRotDiscreteCircle` and make sure they wiork

Fixes #28530

---

#### Review [... cut]
closed 2021-02-25 by peterfpeterson