mantidproject/mantid #28538: Unable to import simpleapi on mac

gvardany, 2020-04-21:
With an installed version of MantidPlot or Workbench, it is impossible to import mantid from python on macOS. It throws the following error:

[image]

gvardany, 2020-04-22:
```
$ python3
Python 3.7.4 (default, Jul  9 2019, 18:13:23) 
[Clang 10.0.1 (clang-1001.0.46.4)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from mantid.simpleapi import *
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Applications/MantidPlot.app/Contents/MacOS/mantid/__init__.py", line 106, in <module>
    from mantid import kernel as _kernel
  File "/Applications/MantidPlot.app/Contents/MacOS/mantid/kernel/__init__.py", line 38, in <module>
    import_mantid_cext('._kernel', 'mantid.kernel', globals())
  File "/Applications/MantidPlot.app/Contents/MacOS/mantid/utils.py", line 35, in import_mantid_cext
    lib = import_module(modulename, package)
  File "/usr/local/Cellar/python/3.7.4/Frameworks/Python.framework/Versions/3.7/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
SystemError: initialization of _kernel raised unreported exception
```

gvardany, 2020-04-22:
From another report, it looks like this is not mac only problem:

```
Traceback (most recent call last):
  File "C:\MantidNightlyInstall\bin\mantid\utils.py", line 35, in
import_mantid_cext
    lib = import_module(modulename, package)
  File
"C:\Users\giull\AppData\Local\Programs\Python\Python38\lib\importlib\__init__.py",
line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in
_find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 657, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 556, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1101, in
create_module
  File "<frozen importlib._bootstrap>", line 219, in
_call_with_frames_removed
ImportError: DLL load failed while importing _kernel: The specified
module could not be found.
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "C:\MantidNightlyInstall\bin\mantid\utils.py", line 39, in
import_mantid_cext
    lib = import_module(modulename.lstrip('.'))
  File
"C:\Users\giull\AppData\Local\Programs\Python\Python38\lib\importlib\__init__.py",
line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 973, in
_find_and_load_unlocked
ModuleNotFoundError: No module named '_kernel'
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "<input>", line 4, in <module>
  File "C:\Program Files\JetBrains\PyCharm Community Edition
2020.1\plugins\python-ce\helpers\pydev\_pydev_bundle\pydev_import_hook.py",
line 21, in do_import
    module = self._system_import(name, *args, **kwargs)
  File "C:\MantidNightlyInstall\bin\mantid\__init__.py", line 105, in
<module>
    from mantid import kernel as _kernel
  File "C:\Program Files\JetBrains\PyCharm Community Edition
2020.1\plugins\python-ce\helpers\pydev\_pydev_bundle\pydev_import_hook.py",
line 21, in do_import
    module = self._system_import(name, *args, **kwargs)
  File "C:\MantidNightlyInstall\bin\mantid\kernel\__init__.py", line
36, in <module>
    import_mantid_cext('._kernel', 'mantid.kernel', globals())
  File "C:\MantidNightlyInstall\bin\mantid\utils.py", line 44, in
import_mantid_cext
    raise ImportError(msg)
ImportError: First import of "._kernel" failed with "DLL load failed
while importing _kernel: The specified module could not be found.".
Second import of "_kernel" failed with "No module named '_kernel'". 
```

gvardany, 2020-04-23:
Apparently, when using the python bundled with Mantid, it works.
However, this could and should be made to work also with the system python, as long as it is at compatible version.

JensenKaplan, 2020-05-21:
Has there been any update to this issue?

martyngigg, 2020-11-03:
What would we mean by system Python on Mac/Windows? This should already work on Linux as we build against that version. For macOS/Windows it is much more uncertain what Python is going to be available so offering general support more challengning. I think our best bet is to use Conda. The nightly packages for Linux are available at: https://anaconda.org/mantid/mantid-framework/files.

JackEAllen, 2021-07-14:
Any update on this issue? I'm also experiencing the same problem on a Windows 7 machine when importing Mantid in IBEX using a version of Python3 not bundled with Mantid on NDXIRIS.

closed 2022-01-17 by stale[bot]