mantidproject/mantid #28544: Add Automated Test for launching Mantid

DanielMurphy22, 2020-04-21:
In line with #28543, running a test which simply checks that Mantid can open and close correctly would be very valuable.

DanielMurphy22, 2020-04-23:
We should look at adding some sort of startup test for Workbench in the system tests. We'll just need to make sure we can easily disable them for framework only builds.

_Originally posted by @martyngigg in https://github.com/mantidproject/mantid/pull/28048#issuecomment-590387119_

martyngigg, 2020-09-10:
We currently have [GUIStartupTest.py](https://github.com/mantidproject/mantid/blob/master/Testing/SystemTests/tests/analysis/GUIStartupTest.py). This needs to be modified, renamed to `WorkbenchStartupTest.py` and be able to function on each platform.

PR #29627 (2020-10-19): Add System Test for Launching Workbench
**Description of work.**
This PR introduces a system test for testing that mantid workbench opens and closes ok for Linux, MacOS and Windows.

**To test:**
Make sure the new system test passes for all platforms, and that it is a good test.

Fixes #28544

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code conform to the [coding standards](http://developer.mantidproject.org/Standards/)?
- Ar [... cut]
closed 2020-10-19 by gvardany