mantidproject/mantid #28556: Investigate background variations when merging MD workspaces

RichardWaiteSTFC, 2020-04-22:
Merging MD workspaces with different orientations produces anomalous backgrounds in overlapping regions of reciprocal space. This is because the background is lambda dependant and the runs may have different statistics in that region (either because of counting time or incident flux). For more information see [1]

there exist algorithms to handle this for TOPAZ (possibly MDNormSCD? Need to confirm with Pas) but these apparently do not work on WISH data.

Need to 
(1) Reproduce the anomalies seen in merged WISH data (Pas is trying to contact a student with some good test data)
(2) See if there is an obvious reason (e.g. event data assumed) that would mean the algorithm does not work for WISH data

[1] http://journals.iucr.org/j/issues/2016/02/00/fs5119/fs5119.pdf

RichardWaiteSTFC, 2020-11-13:
The SingleCrystalDiffuseReduction works well on WISH data if tube ends are masked (responsible for streaks of high intensity) and a background run is used (in this case generated using EnggEstimateFocussedBackground - which took 2 hours for a run and still has some artifacts). I believe the variations in the background were caused by the scattering angle dependent background in the raw data, not the algorithm normalisation. The workflow is slightly clunky given the data need to be loaded, converted to event data and then  saved again - perhaps the algorithm can also take in workspaces as well as nexus files - and convert to event if necessary.
[image]

closed 2020-11-13 by RichardWaiteSTFC

AndreiSavici, 2020-11-13:
@RichardWaiteSTFC The algorithm to use is MDNorm (cleaned up version of MDNormSCD and merged with MDNormDirectSC)

RichardWaiteSTFC, 2020-11-13:
> MDNorm

Thanks @AndreiSavici - I see that this allows for the workspace to be passed instead of a filename which could streamline the workflow a little (although SingleCrystalDiffuseReduction.py looks like it does a few other useful things to prepare the data for MDNorm). As you are more familiar with the algorithms, do you know of any reason why the data has to be event?

AndreiSavici, 2020-11-17:
Data for MDNorm is MDEvents, the output from ConvertToMD. An MDEvent can be a histogram bin. SingleCrystalDiffuseReduction is meant to be like a workflow in a black box, and was developed to deal with data from TOPAZ and CORELLI, so there might be some assumptions made.