mantidproject/mantid #28557: Investigate sample centering corrections when finding UB matrix for WISH

RichardWaiteSTFC, 2020-04-22:
When a sample is off the centre of rotation and beam then there will be a systematic effect on the d-spacing of reflections that will depend on the scattering angle. Ultimately such centering can be accounted for/optimised if the symmetrically equivalent reflections with the same d-spacing) are observed at many angles (as is done on SXD and TOPAZ). However, the detector coverage of WISH is not sufficient to do this with one run, therefore it is necessary to optimise lattice parameter, UB and sample placement for runs at different goniometer angles. 

Need to investigate:
(1) What information is required to use existing algorithms (ask Pas for name), is there currently a way to provide this from what we can work out from the runs individually?
(2) If not, what workflow/algorithms are necessary to optimise the UB and sample placement across runs with different goniometer angles.

Report back to Pas (WISH) before setting up a meeting with TOPAZ and SXD instrument scientists.

DannyHindson, 2020-05-01:
The OptimizeCrystalPlacement algorithm may do what's required here. Will take a look at what this does. 

The example given in the Mantid documentation shows this algorithm being run twice on the same input peaks workspace: once with AdjustSampleOffsets=false and then with AdjustSampleOffsets=true. I think this results in the first run optimizing the sample orientation incorporating the "per run" goniometer data and the second run then optimizes the sample offsets (without any "per run" data being input into the fit)

There is also a python algorithm called OptimizeCrystalPlacementByRun that calls OptimizeCrystalPlacement several times (each time with AdjustSampleOffsets hardcoded to be true). Haven't worked out what the use case for this wrapper algorithm is yet

DannyHindson, 2020-06-11:
I've written the following script that runs OptimizeCrystalPlacement on a set of WISH runs. 

[OptimizeCrystalPlacement.zip](https://github.com/mantidproject/mantid/files/4763794/OptimizeCrystalPlacement.zip)

The script takes these inputs for each run: a UB matrix in the form of a .mat file and a nexus file. The UB matrices have the goniometer rotations for the run embedded in them so the script backs these out based on the data in the sample logs.

Here are the results (in metres):
| Run Number                                                                          | X offset | X offset err | Y offset | Y offset err | Z offset | Z offset err |
|-------------------------------------------------------------------------------------|----------|--------------|----------|--------------|----------|--------------|
| 42,716                                                                              | -0.0030  | 0.0025       | -0.0022  | 0.0015       | 0.0033   | 0.0019       |
| 42,728                                                                              | -0.0015  | 0.0018       | -0.0021  | 0.0011       | 0.0008   | 0.0014       |
| 42,730                                                                              | 0.0007   | 0.0017       | -0.0009  | 0.0012       | 0.0012   | 0.0016       |
| 42,731                                                                              | 0.0007   | 0.0023       | -0.0026  | 0.0015       | -0.0044  | 0.0019       |
| 42,732                                                                              | 0.0005   | 0.0019       | -0.0014  | 0.0012       | -0.0001  | 0.0015       |
| 42,733                                                                              | 0.0013   | 0.0015       | -0.0026  | 0.0010       | -0.0006  | 0.0012       |
| 42,734                                                                              | 0.0002   | 0.0020       | -0.0032  | 0.0013       | -0.0032  | 0.0016       |
|                                                                                     |          |              |          |              |          |              |
| Averages                                                                            | -0.0002  | 0.0020       | -0.0021  | 0.0013       | -0.0004  | 0.0016       |
|                                                                                     |          |              |          |              |          |              |

Y offset is the only one that's significant at ~2mm (and approx consistent across all runs)

DannyHindson, 2020-06-11:
Possible code changes to improve use of OptimizeCrystalPlacement:
- Goniometer::recalculateR should perhaps manage goniometers better where one of the axes isn't equal to x, y or z eg WISH's goniometer only has the phi (EWALD_Pos) and omega (CCR_Pos) degrees of freedom with a fixed chi=45 degrees. So should be possible to set up two axes in SetGoniometer as follows:
SetGoniometer(Workspace=WISHFileName,Axis0='CCR_Pos,0,1,0,1',Axis1='EWALD_Pos,0,1,1,-1')
... but that doesn't work as the implicit chi rotation isn't applied
- documentation could be usefully updated to indicate that any results relating to the goniometer rotations will be presented as if the instrument has a universal goniometer regardless of initial goniometer motors set up
- no way at the moment of constraining one of the goniometer angles to be fixed across runs. Because the actual goniometer gets converted into an equivalent universal goniometer inside OptimizeCrystalPlacement it is assumed all 3 angles are free to move

DannyHindson, 2020-06-30:
Closing this issue because initial investigation has been carried out. Will report back to Pas at next meeting with WISH team

closed 2020-06-30 by DannyHindson