mantidproject/mantid #28569: Problems with the workbench log viewer

NickDraper, 2020-04-23:
1. Does it display filtered or all of the data?
1. We should have an option to unfilter the data
1. If you move down the table with the down key then the graph and stats do not update
1. duration is 0 based to the start of the file, not the start of the run (compare to mantidplot)

NickDraper, 2020-04-23:
Handy file for testing multi-period data:
- POLREF00004699.nxs

NickDraper, 2020-04-29:
> If you could add this, would be awesome:
> 5) Collapse the plot tab if the workspace is not event and the log is not time series.

@gvardany Can I just check what you are after here?  Do you want:
1. The whole stats and graph section to be hidden if none of the logs are time series
1. The graph to disappear and the stats section expand if a selected log is not time series
1. Both the graph and stats section to disappear if the selected log is not time series

NickDraper, 2020-04-30:
Note to self:
For colouring the cells of the logs table
``` python
from qtpy.QtGui import QColor
from qtpy.QtCore import Qt

            item.setData(QColor(Qt.red), Qt.BackgroundRole)
            item.setData(QColor(Qt.blue), Qt.ForegroundRole)

#or from the model
model.setData(model.index(0,0), QColor(Qt.red), Qt.BackgroundRole)
```

NickDraper, 2020-05-01:
@gvardany 
Looking at this file as an example UnitTest/ILL/D22/192068.nxs
I think that option 1 would make the most sense.
[image]

gvardany, 2020-05-04:
And what if just one log is time series? Will they reappear when selecting that log? (D22 does not scan but the scanning detectors have one or two series logs).

NickDraper, 2020-05-04:
> And what if just one log is time series? Will they reappear when selecting that log? (D22 does not scan but the scanning detectors have one or two series logs).

Take a look in #28634, I've done it at the moment that the presence of the stats and graph section will disappear if none of the logs would benefit from it, so it only one or two will it will be present.

closed 2020-05-28 by NickDraper