mantidproject/mantid #28586: Update SaveNXcanSAS & LoadNXcanSAS to NXcanSAS v1.1

smk78, 2020-04-24:
### Nature of the issue
The NXcanSAS v1.0 definition requires a field called _radiation_ in its use of the (optional) NeXus NXsource group. No such field exists in the NXsource definition, although there is a field _type_ which provides the same information.

Since there exist data files written to the NXcanSAS v1.0 specification, it is too late to change the NXcanSAS definition.

### Agreed Resolution
It is agreed to mark the _radiation_ field as deprecated.

This paragraph is to be added to the NXcanSAS documentation for the radiation field:

> This field contains a value from either the NXsource/probe or NXsource/type field. Thus, it is redundant with existing NeXus structure.

Existing data files with the _radiation_ field will not be changed but data files written after this revision should use the _probe_ and/or _type_ fields instead.

This change will be formally identified as NXcanSAS v1.1

### Further Information
https://github.com/nexusformat/definitions/issues/765

smk78, 2020-04-30:
Attachment includes a 1D and 2D example of data written in NXcanSAS v1.1 (generated by SasView developer build of ticket1527 branch).

[sasview-nxcansas-v1p1-examples.zip](https://github.com/mantidproject/mantid/files/4557995/sasview-nxcansas-v1p1-examples.zip)

gemmaguest, 2020-07-29:
Related to #18336

gemmaguest, 2020-08-03:
We will drop support for writing v1.0 but keep ability to read both

PR #29152 (2020-08-11): Move to NXcanSAS v1.1 for saving of radiation type
**Description of work.**

This moves the NXcanSAS processing forward from v1.0 to v1.1.  The major changes between the versions is that

1. The radiation source type is now stored in `/sasentry/sasinstrument/sassource/type` instead of `/sasentry/sasinstrument/sassource/radiation`, which has now been deprecated.
2. The version number has been incremented to v1.1

Our loading of NXcanSAS files ignores the sasinstrument beyond retrieving the name for loading the IDF, so there were no changes needed to be made to the file loader.

**To test:**

1. [... cut]
closed 2020-08-11 by gvardany