mantidproject/mantid #28608: Colorfill plot: x label cut off

DanielMurphy22, 2020-04-28:
**Original reporter:** SANS group / rob Dalgliesh

### Expected behavior
WB plots should open well scaled
### Actual behavior
This can be hard with mpl
The x label was cut off in Windows 10 for the label q(A^-1)
### Steps to reproduce the behavior
I think this is most liekly to find for colorfill plots on Windows
### Platforms affected
WB 5.0

NickDraper, 2020-05-29:
@DanielMurphy22 I'm having trouble reproducing this with the current code, could you have a go as well and see if you can come up with an example?

DanielMurphy22, 2020-05-29:
> @DanielMurphy22 I'm having trouble reproducing this with the current code, could you have a go as well and see if you can come up with an example?

I've had a quick go with no luck. I wonder if this depends on the size of your screen. I had thought 3D plots would be at more risk for this sort of thing but the one's I've checked are fine. Colorfill plots still have the diagonal axis numbers which I remember they disliked.

NickDraper, 2020-05-29:
Yes, the diagonal labels are on a different issue.  I even tried scaling up the values in the X axis but still could not reproduice this.
[image]
[image]

NickDraper, 2020-05-29:
Ah, it looks like there is a sweet spot, 
[image]
but this would be solved by the labels being horizontal.

NickDraper, 2020-05-29:
Should be solved by #28601

closed 2020-05-29 by NickDraper