mantidproject/mantid #28620: Workench sample logs: Unicode warnings on reading units

NickDraper, 2020-04-29:


NickDraper, 2020-04-29:
@martyngigg Any ideas?

martyngigg, 2020-04-29:
This is because we have a byte string that is being converted to unicode without being told what encoding to use so it has defaulted to `utf-8` but doesn't know how to decode that byte. We need to find out from Freddie what encoding is used in those logs/unit strings.

NickDraper, 2020-04-30:
The Nexus files calls it ascii

martyngigg, 2020-04-30:
I now remember that I had an email conversation with Freddie about this but it didn't make it to an issue as he his going to make sure future data is encoded in utf8:

> 
> Hi,
>  
> NeXus prefers utf8 but doesn’t require it https://manual.nexusformat.org/nxdl-types.html however it also does not say how to indicate if it is not utf8! HDF5 has support for utf8 https://support.hdfgroup.org/HDF5/doc/Advanced/UsingUnicode/index.html but i will currently be writing everything as ascii. Given the mu is encoded in one byte I am guessing it was originally Windows 1252 but this happens to agree with UTF16 for a mu.
>  
> So I will need to make sure I convert character data to UTF8   
>  
> Regards,
>  
> Freddie
>

The only think I can think of for old data is to have a list of fallback encodings we can try. I think we would need to look in `https://github.com/mantidproject/mantid/blob/master/Framework/PythonInterface/mantid/kernel/src/Exports/Property.cpp#L109` and replace 
```make_function(&Property::units,
                 return_value_policy<copy_const_reference>())
```
with a free function but I'm not sure what that wold look like. Would you like me to take a look at this?

NickDraper, 2020-05-01:
solved with #28623

closed 2020-05-01 by NickDraper