mantidproject/mantid #28621: Crash closing plot window while context menu is showing

gemmaguest, 2020-04-30:
### Expected behavior
Context menu should be cleaned up when plot window is closed

### Actual behavior
The context menu remains visible and Mantid crashes

### Steps to reproduce the behavior
- Open Workbench
- Load and plot some data e.g. `ws=Load('INTER13460')`
- Right-click in the plot window to show the context menu
- Click the X to close the plot window. The context menu remains visible.
- Click anywhere Workbench to dismiss the context menu. Mantid crashes.

```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/matplotlib/cbook/__init__.py", line 388, in process
    proxy(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/matplotlib/cbook/__init__.py", line 228, in __call__
    return mtd(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/matplotlib/pyplot.py", line 547, in make_active
    _pylab_helpers.Gcf.set_active(figManager)
  File "/home/gemma/work/checkout/mantid/qt/applications/workbench/workbench/plotting/globalfiguremanager.py", line 189, in set_active
    cls.figs[manager.num] = manager
  File "/home/gemma/work/checkout/mantid/qt/applications/workbench/workbench/plotting/observabledictionary.py", line 50, in __setitem__
    self._notify_observers(action, key)
  File "/home/gemma/work/checkout/mantid/qt/applications/workbench/workbench/plotting/observabledictionary.py", line 41, in _notify_observers
    observer.notify(action, key)
  File "/home/gemma/work/checkout/mantid/qt/applications/workbench/workbench/plotting/globalfiguremanager.py", line 47, in notify
    self.figure_manager.notify_observers(FigureAction.New, key)
  File "/home/gemma/work/checkout/mantid/qt/applications/workbench/workbench/plotting/globalfiguremanager.py", line 257, in notify_observers
    observer.notify(action, figure_number)
  File "/home/gemma/work/checkout/mantid/qt/applications/workbench/workbench/widgets/plotselector/model.py", line 72, in notify
    self.presenter.append_to_plot_list(plot_number)
  File "/home/gemma/work/checkout/mantid/qt/applications/workbench/workbench/widgets/plotselector/presenter.py", line 65, in append_to_plot_list
    self.view.append_to_plot_list(plot_number)
  File "/home/gemma/work/checkout/mantid/qt/python/mantidqt/utils/qt/qappthreadcall.py", line 46, in __call__
    return self.callee(*args, **kwargs)
  File "/home/gemma/work/checkout/mantid/qt/applications/workbench/workbench/widgets/plotselector/view.py", line 226, in append_to_plot_list
    plot_name_widget = PlotNameWidget(self.presenter, plot_number, self)
  File "/home/gemma/work/checkout/mantid/qt/applications/workbench/workbench/widgets/plotselector/view.py", line 616, in __init__
    self.line_edit = QLineEdit(self.presenter.get_plot_name_from_number(plot_number))
  File "/home/gemma/work/checkout/mantid/qt/applications/workbench/workbench/widgets/plotselector/presenter.py", line 48, in get_plot_name_from_number
    return self.model.get_plot_name_from_number(plot_number)
  File "/home/gemma/work/checkout/mantid/qt/applications/workbench/workbench/widgets/plotselector/model.py", line 39, in get_plot_name_from_number
    return figure_manager.get_window_title()
  File "/home/gemma/work/checkout/mantid/qt/applications/workbench/workbench/plotting/figuremanager.py", line 375, in get_window_title
    return self.window.windowTitle()
RuntimeError: wrapped C/C++ object of type FigureWindow has been deleted
Segmentation fault (core dumped)
```

### Platforms affected
Ubuntu 18.04, maybe all

NickDraper, 2020-05-06:
On windows the context menu is automatically closed by windows whenever you click off it (i.e. to click the X box)  so this cannot happen on windows.

ConorMFinn, 2020-05-18:
I am unable to recreate this on Ubuntu 18.04 as well. Clicking on the X button simply closes the context menu for me, and does not close the figure window until the context menu is gone. I'm using the GNOME desktop environment, if it helps.

NickDraper, 2020-05-18:
@gemmaguest Can you still recreate this?

gemmaguest, 2020-05-18:
Ah I am using X11 forwarding from linux to an Xming server on Windows. So maybe it's an Xming thing. Sorry, probably quite a niche use case!

closed 2021-09-07 by stale[bot]