mantidproject/mantid #28633: Plots Created From Sample Logs Cannot Open Figure Options

ConorMFinn, 2020-05-01:
### Expected behavior
The figure options should open correctly. 
### Actual behavior
An unhandled exception occurs. 
```
Traceback (most recent call last):
  File "/opt/mantidnightly/lib/workbench/plotting/figuremanager.py", line 346, in launch_plot_options
    self.plot_options_dialog = PlotConfigDialogPresenter(self.canvas.figure, parent=self.window)
  File "/opt/mantidnightly/lib/mantidqt/widgets/plotconfigdialog/presenter.py", line 42, in __init__
    curves_tab = CurvesTabWidgetPresenter(self.fig, parent=self.view, legend_tab=legend_tab)
  File "/opt/mantidnightly/lib/mantidqt/widgets/plotconfigdialog/curvestabwidget/presenter.py", line 42, in __init__
    self.populate_curve_combo_box_and_update_view()
  File "/opt/mantidnightly/lib/mantidqt/widgets/plotconfigdialog/curvestabwidget/presenter.py", line 166, in populate_curve_combo_box_and_update_view
    self.update_view()
  File "/opt/mantidnightly/lib/mantidqt/widgets/plotconfigdialog/curvestabwidget/presenter.py", line 263, in update_view
    self.set_errorbars_tab_enabled()
  File "/opt/mantidnightly/lib/mantidqt/widgets/plotconfigdialog/curvestabwidget/presenter.py", line 256, in set_errorbars_tab_enabled
    enable_errorbars = curve_has_errors(self.get_selected_curve())
  File "/opt/mantidnightly/lib/mantidqt/widgets/plotconfigdialog/curvestabwidget/__init__.py", line 87, in curve_has_errors
    return ax.artists_workspace_has_errors(curve)
  File "/opt/mantidnightly/lib/mantid/plots/mantidaxes.py", line 271, in artists_workspace_has_errors
    workspace_index = workspace.getIndexFromSpectrumNumber(spec_num)
Boost.Python.ArgumentError: Python argument types in
    MatrixWorkspace.getIndexFromSpectrumNumber(Workspace2D, NoneType)
did not match C++ signature:
    getIndexFromSpectrumNumber(Mantid::API::MatrixWorkspace {lvalue} self, int spec_no)
```
### Steps to reproduce the behavior
1. Load something like POLREF14966 with sample logs. 
2. Right click the workspace and select "Show Sample Logs"
3. Select a sample log with a plot, such as `good_uah_log`
4. Double click the plot to open a plotting window. 
5. Attempt to open the figure options by clicking the cog. 

### Platforms affected
All? Found on Ubuntu 18.04.

PhilColebrooke, 2020-05-04:
#27770

ConorMFinn, 2020-05-04:
Closed as duplicate

closed 2020-05-04 by ConorMFinn