mantidproject/mantid #28638: Unrolling RROA history with polynomial correction fails

gemmaguest, 2020-05-04:
### Expected behavior
Re-running unrolled history should run the same reduction successfully

### Actual behavior
There is an error running the polynomial correction step

### Steps to reproduce the behavior
- Run this script
```
Load('INTER13460', OutputWorkspace='TOF_13460')
ReflectometryReductionOneAuto(InputWorkspace='TOF_13460', ThetaIn=0.5)
```
- Show history for `IvsQ_13460` and tick unroll for `ReflectometryReductionOneAuto` and `ReflectometryReductionOne`
- Save or copy it into a script.
- Running that script gives an error on the polynomial correction step.

### Notes
- Note that running `ReflectometryReductionOne` directly does not exhibit this problem so it is something to do with the way `ReflectometryReductionOneAuto` is running `ReflectometryReductionOne`.
- This is the unrolled history. The problem is that `OutputWorkspace` is set to an empty string in the `PolynomialCorrection` algorithm.
```
Load(Filename='/home/gemma/work/checkout/mantid/build/ExternalData/Testing/Data/UnitTest/INTER00013460.nxs', OutputWorkspace='TOF_13460')

# Child algorithms of ReflectometryReductionOneAuto
SpecularReflectionPositionCorrect(InputWorkspace='TOF_13460', TwoTheta=1, DetectorComponentName='point-detector', OutputWorkspace='__TMP0x7f78185374a0')

## Child algorithms of ReflectometryReductionOne
GroupDetectors(InputWorkspace='__TMP0x7f78185374a0', OutputWorkspace='__TMP0x7f781860b860', GroupingPattern='3')
ConvertUnits(InputWorkspace='__TMP0x7f781860b860', OutputWorkspace='__TMP0x7f78180c2300', Target='Wavelength', AlignBins=True)
CropWorkspace(InputWorkspace='__TMP0x7f78185374a0', OutputWorkspace='__TMP0x7f7818041490', StartWorkspaceIndex=2, EndWorkspaceIndex=2)
ConvertUnits(InputWorkspace='__TMP0x7f7818041490', OutputWorkspace='__TMP0x7f781815def0', Target='Wavelength', AlignBins=True)
CalculateFlatBackground(InputWorkspace='__TMP0x7f781815def0', OutputWorkspace='__TMP0x7f78181f2dc0', StartX=17, EndX=18)
RebinToWorkspace(WorkspaceToRebin='__TMP0x7f78180c2300', WorkspaceToMatch='__TMP0x7f78181f2dc0', OutputWorkspace='__TMP0x7f781815def0')
Divide(LHSWorkspace='__TMP0x7f781815def0', RHSWorkspace='__TMP0x7f78181f2dc0', OutputWorkspace='__TMP0x7f78183c5b60')
CropWorkspace(InputWorkspace='__TMP0x7f78183c5b60', OutputWorkspace='__TMP0x7f78180c2300', XMin=1.5, XMax=17)
PolynomialCorrection(InputWorkspace='__TMP0x7f78180c2300', OutputWorkspace='', Coefficients='35.5893,-24.5591,9.20375,-1.89265,0.222291,-0.0148746,0.00052709,-7.66807e-06', Operation='Divide')
ConvertUnits(InputWorkspace='__TMP0x7f78183c5b60', OutputWorkspace='IvsQ_13460', Target='MomentumTransfer')
## End of child algorithms of ReflectometryReductionOne

NRCalculateSlitResolution(Workspace='__TMP0x7f78180c2300', TwoTheta=1)
Rebin(InputWorkspace='__TMP0x7f78180c2300', OutputWorkspace='IvsQ_binned_13460', Params='0.00645279,-0.047641,0.072643')
# End of child algorithms of ReflectometryReductionOneAuto
```

### Platforms affected
All

gemmaguest, 2020-05-04:
`ReflectometryReductionOne2::algorithmicCorrection` runs `PolynomialCorrection` as a child algorithm. It does not set an output workspace name for the child algorithm. For some reason this causes the output workspace to be set to an empty string in our history instead of the temporary (`__TMP`) string like all of the other child algorithms (which we run in the same way). I wonder if this is because it is a `UnaryOperation`.

It is odd that this works from `RRO` directly but not when `RRO` is a child of `RROA`. 

This needs some more investigation although I'm not really sure where to start, but maybe:
- We could check if there are other UnaryOperations run as a child without an output name being set (either from other algorithms or tests).
- We could take a look at the way the history is generated and see where the `__TMP` names get added and try to find why this is not done for the UnaryOperation.

NickDraper, 2020-05-04:
Try setting a name in the property for the output workspace for the polynomial correction after line 635 in RRO2.cpp
```
  corrAlg->setPropertyValue("OutputWorkspace", "Dummy_WS_NAME");
```
See if that helps (No promises!)

gemmaguest, 2020-05-05:
After playing around a bit more I think this is related to the cloned workspace we create in `convertToQ` when summing in lambda. When summing in Q it works ok. I suggest we wait till that is moved to a proper algorithm (as per #23012) and then check if it fixes this.

gemmaguest, 2020-07-30:
This is also a problem with transmission correction or no correction. I think it is because when !debug then RROA does not have a name for the output IvsLam workspace and somehow this empty name is perhaps getting used in RRO, instead of a temp name, although I can't see where.

gemmaguest, 2021-04-15:
There is a similar issue with the output of CreateTransmissionWorkspace having an empty name when run as a child of ReflectometryReductionOne. ReflectometryReductionOne successfully uses it in the next step internally, but the temporary name that appears in the unrolled history does not get created anywhere earlier in the history - the output where it should appear has an empty name. 

I thought this might be to do with the fact that we set a default name for this output if one is not given and that somehow that was overriding the temporary name, but I've experimented with removing that and this is how I discovered that the output name is empty otherwise in that case.

closed 2023-05-12 by stale[bot]